tuxedocomputers/tuxedo-keyboard #115: Mo modes availabel on XMG APEX15

3isenHeiM, 2021-11-16:
Thanks for this tool, I was able to install it on my XMG APEX15 :)

However using the ``Fn`` + ``/`` to change mode, I only this sequernce : RED, GREEN, BLUE, YELLOW, PURPLE, CYAN, WHITE, OFF, and these are only static colors.

**I can't set the Breathe mode, or any dynamic one.**

Here is the log if the ``npm install`` command :

```

> tuxedo-control-center@1.1.0 install
> node-gyp rebuild

gyp info it worked if it ends with ok
gyp info using node-gyp@7.1.2
gyp info using node@12.22.5 | linux | x64
gyp info find Python using Python version 3.9.2 found at "/usr/bin/python3"
gyp info spawn /usr/bin/python3
gyp info spawn args [
gyp info spawn args   '/usr/share/nodejs/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/home/pierre/Documents/GitHub/tuxedo-control-center/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/share/nodejs/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/include/nodejs/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/usr/include/nodejs',
gyp info spawn args   '-Dnode_gyp_dir=/usr/share/nodejs/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/usr/include/nodejs/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/home/pierre/Documents/GitHub/tuxedo-control-center',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory '/home/pierre/Documents/GitHub/tuxedo-control-center/build'
  CC(target) Release/obj.target/nothing/node_modules/node-addon-api/src/nothing.o
  AR(target) Release/obj.target/node_modules/node-addon-api/src/nothing.a
  COPY Release/nothing.a
  CXX(target) Release/obj.target/TuxedoIOAPI/src/native-lib/tuxedo_io_napi.o
  SOLINK_MODULE(target) Release/obj.target/TuxedoIOAPI.node
  COPY Release/TuxedoIOAPI.node
make: Leaving directory '/home/pierre/Documents/GitHub/tuxedo-control-center/build'
gyp info ok 

up to date, audited 1381 packages in 9s

69 vulnerabilities (1 low, 32 moderate, 34 high, 2 critical)

To address issues that do not require attention, run:
  npm audit fix

To address all issues possible (including breaking changes), run:
  npm audit fix --force

Some issues need review, and may require choosing
a different dependency.

Run `npm audit` for details.
```

Where have I failed ?

tuxedoxt, 2021-11-16:
Hello,

off hand I do not know what modes are available for that keyboard. In general the Fn key will just cycle colors even if other modes are supported.

You can try to play around with the sysfs interface in `/sys/devices/platform/tuxedo_keyboard/`.

Specifically https://github.com/tuxedocomputers/tuxedo-keyboard#mode would be relevant

3isenHeiM, 2021-11-19:
Thanks.

I don't find the method on how to play with these mode.

Entering the command `sudo modprobe tuxedo_keyboard mode=7` for instance didn't help...

tuxedoxt, 2021-11-19:
You can write to the sysfs "files" directly, for example...

```
echo 7 | sudo tee /sys/devices/platform/tuxedo_keyboard/mode
```

3isenHeiM, 2021-11-19:
Seems that not every feature is available on my model, I cannot have dynamic mode, not left-right colors ;)

Thanks for the help anyway !

closed 2021-11-19 by 3isenHeiM