tuxedocomputers/tuxedo-keyboard #116: Not working with Manjaro on Infinitybook 14

petrarca-arezzo, 2021-11-29:
I*m running Manjaro KDE Plasma (stable) on a Infinitybook Pro 14 Gen 6, installed `tuxedo-keyboard` from the repository (cf. screenshot) but can*t get it working:
```
sudo modprobe tuxedo_keyboard
modprobe: FATAL: Module tuxedo_keyboard not found in directory /lib/modules/5.13.19-2-MANJARO
```
[image]

tuxedoxt, 2021-11-29:
Hello, please check the output of

```
dkms status
```

it needs to say installed for the booted kernel (`uname -r`).

If it doesn't you can try to manually build/install it with

```
sudo dkms install tuxedo-keyboard/3.0.9
```

or refer to any errors generated.

George4All, 2021-11-30:
I've had a similar issue on elementary OS 6.0. Running 
`sudo make dkmsinstall`
returms
`cp -R . /usr/src/tuxedo-keyboard-3.0.9
dkms install -m tuxedo-keyboard -v 3.0.9
Error! echo
Your kernel headers for kernel 5.11.0-40-generic cannot be found at
/lib/modules/5.11.0-40-generic/build or /lib/modules/5.11.0-40-generic/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
make: *** [Makefile:43: dkmsinstall] Error 1
`

petrarca-arezzo, 2021-11-30:
I had to install the linux headers (via pamac) and have the following now:
```
> dkms status
tuxedo-keyboard/3.0.9, 5.13.19-2-MANJARO, x86_64: installed
> uname -r
5.13.19-2-MANJARO
> cat /etc/modprobe.d/tuxedo_keyboard.conf
options tuxedo-keyboard mode=0 brightness=75 color_left=0xFFFFFF color_center=0xFFFFFF color_right=0xFFFFFF
```
Probably, I'm too dumb to see it, but I would I enable the backlight (the TCC doesn't show any option). As mentioned I installed tuxedo_keyboard via Pamac; there is no tuxedo-io, clevo-wmi, or clevo-acpi available via Pamac.

petrarca-arezzo, 2021-12-10:
Any idea what might be the issue (or what I'm overlooking) that I can't enable the keyboard backlight?

tuxedoxt, 2021-12-10:
There is (currently) no special driver for the keyboard backlight for your model. The Fn+space (I believe) should work anyway.

Uhter, 2021-12-10:
Hello,
I'm not an expert but, have you see/try [this](https://wiki.archlinux.org/title/TUXEDO_InfinityBook_Pro_15v4#Function_keys) under 'Function keys' section?

tuxedoxt, 2021-12-10:
> Hello, I'm not an expert but, have you see/try [this](https://wiki.archlinux.org/title/TUXEDO_InfinityBook_Pro_15v4#Function_keys) under 'Function keys' section?

Good point, make sure those `acpi_` parameters are **not** set.

`tuxedo-keyboard` will map the missing keys (like airplane, touchpad)

petrarca-arezzo, 2021-12-10:
Wonderful - in the end just Fn+<space-bar> did the trick !

closed 2021-12-10 by petrarca-arezzo