tuxedocomputers/tuxedo-keyboard #120: How to install on immutable OS (Fedora Silverblue, Kinoite)

ruzko, 2022-02-27:
Due to the nature of these immutable operating systems, kernel modules cannot be built using DKMS. So the question is, how can these modules be installed on OSes with read-only system directories?
I tried cloning into the repo, and running `make clean && make`.
This is the terminal output:
```
make -C /lib/modules/5.16.11-200.fc35.x86_64/build M=/var/home/Jacob/git/tuxedo-keyboard clean
make[1]: Entering directory '/usr/src/kernels/5.16.11-200.fc35.x86_64'
  CLEAN   /var/home/Jacob/git/tuxedo-keyboard/Module.symvers
make[1]: Leaving directory '/usr/src/kernels/5.16.11-200.fc35.x86_64'
make -C /lib/modules/5.16.11-200.fc35.x86_64/build M=/var/home/Jacob/git/tuxedo-keyboard modules
make[1]: Entering directory '/usr/src/kernels/5.16.11-200.fc35.x86_64'
  CC [M]  /var/home/Jacob/git/tuxedo-keyboard/./src/tuxedo_keyboard.o
  CC [M]  /var/home/Jacob/git/tuxedo-keyboard/./src/clevo_wmi.o
  CC [M]  /var/home/Jacob/git/tuxedo-keyboard/./src/clevo_acpi.o
  CC [M]  /var/home/Jacob/git/tuxedo-keyboard/./src/tuxedo_io/tuxedo_io.o
  CC [M]  /var/home/Jacob/git/tuxedo-keyboard/./src/uniwill_wmi.o
  MODPOST /var/home/Jacob/git/tuxedo-keyboard/Module.symvers
  CC [M]  /var/home/Jacob/git/tuxedo-keyboard/./src/clevo_acpi.mod.o
  LD [M]  /var/home/Jacob/git/tuxedo-keyboard/./src/clevo_acpi.ko
  BTF [M] /var/home/Jacob/git/tuxedo-keyboard/./src/clevo_acpi.ko
Skipping BTF generation for /var/home/Jacob/git/tuxedo-keyboard/./src/clevo_acpi.ko due to unavailability of vmlinux
  CC [M]  /var/home/Jacob/git/tuxedo-keyboard/./src/clevo_wmi.mod.o
  LD [M]  /var/home/Jacob/git/tuxedo-keyboard/./src/clevo_wmi.ko
  BTF [M] /var/home/Jacob/git/tuxedo-keyboard/./src/clevo_wmi.ko
Skipping BTF generation for /var/home/Jacob/git/tuxedo-keyboard/./src/clevo_wmi.ko due to unavailability of vmlinux
  CC [M]  /var/home/Jacob/git/tuxedo-keyboard/./src/tuxedo_io/tuxedo_io.mod.o
  LD [M]  /var/home/Jacob/git/tuxedo-keyboard/./src/tuxedo_io/tuxedo_io.ko
  BTF [M] /var/home/Jacob/git/tuxedo-keyboard/./src/tuxedo_io/tuxedo_io.ko
Skipping BTF generation for /var/home/Jacob/git/tuxedo-keyboard/./src/tuxedo_io/tuxedo_io.ko due to unavailability of vmlinux
  CC [M]  /var/home/Jacob/git/tuxedo-keyboard/./src/tuxedo_keyboard.mod.o
  LD [M]  /var/home/Jacob/git/tuxedo-keyboard/./src/tuxedo_keyboard.ko
  BTF [M] /var/home/Jacob/git/tuxedo-keyboard/./src/tuxedo_keyboard.ko
Skipping BTF generation for /var/home/Jacob/git/tuxedo-keyboard/./src/tuxedo_keyboard.ko due to unavailability of vmlinux
  CC [M]  /var/home/Jacob/git/tuxedo-keyboard/./src/uniwill_wmi.mod.o
  LD [M]  /var/home/Jacob/git/tuxedo-keyboard/./src/uniwill_wmi.ko
  BTF [M] /var/home/Jacob/git/tuxedo-keyboard/./src/uniwill_wmi.ko
Skipping BTF generation for /var/home/Jacob/git/tuxedo-keyboard/./src/uniwill_wmi.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/kernels/5.16.11-200.fc35.x86_64'
```

Running `sudo modprobe tuxedo_keyboard` results in:

`modprobe: FATAL: Module tuxedo_keyboard not found in directory /lib/modules/5.16.11-200.fc35.x86_64`

System information courtesy of `inxi -Fazy`:
```
System:
  Kernel: 5.16.11-200.fc35.x86_64 x86_64 bits: 64 compiler: gcc
    v: 2.37-10.fc35
    parameters: BOOT_IMAGE=(hd1,gpt2)/ostree/fedora-16d9461909469c44969ff2fa245a8d8e1e03f2f587385d9f39dbfa0416026d57/vmlinuz-5.16.11-200.fc35.x86_64
    rd.luks.uuid=luks-ac2be2a2-3a0d-463c-a10e-db33888ea41d rhgb quiet
    root=UUID=ab093a6e-4634-4015-bcc4-9cd2d90a1c94 rootflags=subvol=root
    ostree=/ostree/boot.1/fedora/16d9461909469c44969ff2fa245a8d8e1e03f2f587385d9f39dbfa0416026d57/0
    rd.driver.blacklist=nouveau modprobe.blacklist=nouveau
    nvidia-drm.modeset=1
  Desktop: KDE Plasma 5.24.2 tk: Qt 5.15.2 wm: kwin_wayland vt: 1 dm: SDDM
    Distro: Fedora release 35 (Thirty Five)
Machine:
  Type: Laptop System: Notebook product: P95xER v: N/A
    serial: <superuser required> Chassis: No Enclosure type: 10
    serial: <superuser required>
  Mobo: Notebook model: P95xER serial: <superuser required>
    UEFI: American Megatrends v: 1.07.15 date: 11/16/2018
Battery:
  ID-1: BAT0 charge: 45.2 Wh (89.7%) condition: 50.4/53.2 Wh (94.8%)
    volts: 16.6 min: 15.2 model: Notebook BAT type: Li-ion serial: <filter>
    status: N/A
CPU:
  Info: model: Intel Core i7-8750H bits: 64 type: MT MCP arch: Coffee Lake
    family: 6 model-id: 0x9E (158) stepping: 0xA (10) microcode: 0xEC
  Topology: cpus: 1x cores: 6 tpc: 2 threads: 12 smt: enabled cache:
    L1: 384 KiB desc: d-6x32 KiB; i-6x32 KiB L2: 1.5 MiB desc: 6x256 KiB
    L3: 9 MiB desc: 1x9 MiB
  Speed (MHz): avg: 800 min/max: 800/4100 scaling: driver: intel_pstate
    governor: powersave cores: 1: 800 2: 800 3: 800 4: 800 5: 800 6: 800 7: 800
    8: 800 9: 800 10: 800 11: 800 12: 800 bogomips: 52799
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
  Vulnerabilities:
  Type: itlb_multihit status: KVM: VMX disabled
  Type: l1tf
    mitigation: PTE Inversion; VMX: conditional cache flushes, SMT vulnerable
  Type: mds mitigation: Clear CPU buffers; SMT vulnerable
  Type: meltdown mitigation: PTI
  Type: spec_store_bypass
    mitigation: Speculative Store Bypass disabled via prctl
  Type: spectre_v1
    mitigation: usercopy/swapgs barriers and __user pointer sanitization
  Type: spectre_v2 mitigation: Full generic retpoline, IBPB: conditional,
    IBRS_FW, STIBP: conditional, RSB filling
  Type: srbds mitigation: Microcode
  Type: tsx_async_abort status: Not affected
Graphics:
  Device-1: Intel CoffeeLake-H GT2 [UHD Graphics 630] vendor: CLEVO/KAPOK
    driver: i915 v: kernel bus-ID: 00:02.0 chip-ID: 8086:3e9b class-ID: 0300
  Device-2: NVIDIA GP104M [GeForce GTX 1070 Mobile] vendor: CLEVO/KAPOK
    driver: nvidia v: 510.47.03 alternate: nouveau,nvidia_drm bus-ID: 01:00.0
    chip-ID: 10de:1ba1 class-ID: 0300
  Device-3: Chicony Chicony USB 2.0 Camera type: USB driver: uvcvideo
    bus-ID: 1-8:3 chip-ID: 04f2:b5a7 class-ID: 0e02
  Display: wayland server: X.Org 1.21.1.4 compositor: kwin_wayland driver:
    loaded: intel,nvidia display-ID: :1 screens: 1
  Screen-1: 0 s-res: 1920x1080 s-dpi: 96 s-size: 508x285mm (20.0x11.2")
    s-diag: 582mm (22.9")
  Monitor-1: XWAYLAND0 res: 1920x1080 hz: 60 dpi: 139
    size: 350x190mm (13.8x7.5") diag: 398mm (15.7")
  OpenGL: renderer: Mesa Intel UHD Graphics 630 (CFL GT2) v: 4.6 Mesa 21.3.6
    direct render: Yes
Audio:
  Device-1: Intel Cannon Lake PCH cAVS vendor: CLEVO/KAPOK
    driver: snd_hda_intel v: kernel alternate: snd_soc_skl,snd_sof_pci_intel_cnl
    bus-ID: 00:1f.3 chip-ID: 8086:a348 class-ID: 0403
  Device-2: NVIDIA GP104 High Definition Audio driver: snd_hda_intel
    v: kernel bus-ID: 01:00.1 chip-ID: 10de:10f0 class-ID: 0403
  Sound Server-1: ALSA v: k5.16.11-200.fc35.x86_64 running: yes
  Sound Server-2: PipeWire v: 0.3.47 running: yes
Network:
  Device-1: Intel Wireless-AC 9260 vendor: Rivet Networks driver: iwlwifi
    v: kernel bus-ID: 03:00.0 chip-ID: 8086:2526 class-ID: 0280
  IF: wlp3s0 state: up mac: <filter>
  Device-2: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet
    vendor: CLEVO/KAPOK driver: r8169 v: kernel port: 3000 bus-ID: 04:00.0
    chip-ID: 10ec:8168 class-ID: 0200
  IF: enp4s0 state: down mac: <filter>
Bluetooth:
  Device-1: Intel Wireless-AC 9260 Bluetooth Adapter type: USB driver: btusb
    v: 0.8 bus-ID: 1-14:5 chip-ID: 8087:0025 class-ID: e001
  Report: rfkill ID: hci0 rfk-id: 0 state: up address: see --recommends
Drives:
  Local Storage: total: 2.05 TiB used: 457.52 GiB (21.8%)
  SMART Message: Unable to run smartctl. Root privileges required.
  ID-1: /dev/nvme0n1 maj-min: 259:0 vendor: Samsung model: SSD 960 EVO 250GB
    size: 232.89 GiB block-size: physical: 512 B logical: 512 B speed: 31.6 Gb/s
    lanes: 4 type: SSD serial: <filter> rev: 3B7QCXE7 temp: 34.9 C scheme: GPT
  ID-2: /dev/sda maj-min: 8:0 vendor: Seagate model: ST2000LX001-1RG174
    size: 1.82 TiB block-size: physical: 4096 B logical: 512 B speed: 6.0 Gb/s
    type: HDD rpm: 5400 serial: <filter> rev: SDM1
Partition:
  ID-1: /boot raw-size: 1024 MiB size: 973.4 MiB (95.06%)
    used: 80.6 MiB (8.3%) fs: ext4 dev: /dev/nvme0n1p2 maj-min: 259:2
  ID-2: /boot/efi raw-size: 600 MiB size: 598.8 MiB (99.80%)
    used: 9.8 MiB (1.6%) fs: vfat dev: /dev/nvme0n1p1 maj-min: 259:1
Swap:
  Kernel: swappiness: 60 (default) cache-pressure: 100 (default)
  ID-1: swap-1 type: zram size: 8 GiB used: 0 KiB (0.0%) priority: 100
    dev: /dev/zram0
Sensors:
  System Temperatures: cpu: 46.0 C pch: 56.0 C mobo: N/A
  Fan Speeds (RPM): N/A
Info:
  Processes: 362 Uptime: 1h 19m wakeups: 1 Memory: 31.2 GiB
  used: 3.89 GiB (12.5%) Init: systemd v: 249 runlevel: 5
  target: graphical.target tool: systemctl Compilers: gcc: 11.2.1 Packages:
  note: see --pkg flatpak: 64 Shell: Zsh v: 5.8.1 running-in: alacritty
  inxi: 3.3.12
```

ruzko, 2022-02-27:
Merging #62 and getting the modules included in mainline would enable Fedora to ship them with the kernel, circumventing the need to build locally.

zladuric, 2022-03-24:
@ruzko did you find a workaround for this yet? Or better, to be more practical: could you list what does and what doesn't work with Silverblue/Kinoite?

ruzko, 2022-03-25:
@zladuric No workaround, but I've accepted my circumstances. Still using Kinoite as my daily OS.
dkms kernel modules can't be installed, so any features relying on such kernel modules are unavailable.
On my computer:
- Keyboard backlight can't be controlled
- As far as I can tell, gpu fan control (when using nvidia proprietary driver) doesn't work either. Discrete gpu fan rpm never goes below a certain threshold (maybe 40%?), which is unbearable in the long run. This caused me to switch to the Nouveau driver, which seems to behave.
- TCC doesn't work

zladuric, 2022-03-26:
Thanks, useful info.

vega-d, 2022-07-30:
The main problem with using tuxedo-keyboard driver on Silverblue and it's sister distros is the fact that DKMS is not supported. kmods, however, work, but since tuxedo-keyboard copr repo (which is otherwise compatible) tries to use DKMS silverblue just spits errors like this if one tries to install: 
```
Jul 30 16:46:52 dart rpm-ostree(tuxedo-keyboard.post)[66463]: Loading new tuxedo-keyboard-3.0.10 DKMS files...
Jul 30 16:46:52 dart rpm-ostree(tuxedo-keyboard.post)[66500]: Error! No write access to DKMS tree at /var/lib/dkms
Jul 30 16:46:52 dart rpm-ostree(tuxedo-keyboard.post)[66460]: Check tccd running status
Jul 30 16:46:52 dart rpm-ostree(tuxedo-keyboard.post)[66501]: rpm-ostree-systemctl: Ignored non-preset command: is-active --quiet tccd.service
Jul 30 16:46:52 dart rpm-ostree(tuxedo-keyboard.post)[66460]: Stop tccd temporarily
Jul 30 16:46:52 dart rpm-ostree(tuxedo-keyboard.post)[66502]: rpm-ostree-systemctl: Ignored non-preset command: stop tccd
Jul 30 16:46:52 dart rpm-ostree(tuxedo-keyboard.post)[66460]: /proc/self/fd/5: line 21: fg: no job control
Jul 30 16:46:52 dart rpm-ostree(tuxedo-keyboard.post)[66460]: (Re)load modules if possible
Jul 30 16:46:52 dart rpm-ostree(tuxedo-keyboard.post)[66460]: Start tccd again
Jul 30 16:46:52 dart rpm-ostree(tuxedo-keyboard.post)[66515]: rpm-ostree-systemctl: Ignored non-preset command: start tccd

```

A relatively obvious solution would be using kmods instead of dkms modules, but I'm not sure how feasible that is.

closed 2023-12-11 by tuxedo-bot

salim-b, 2023-12-11:
> Merging #62 and getting the modules included in mainline would enable Fedora to ship them with the kernel, circumventing the need to build locally.

See https://gitlab.com/tuxedocomputers/development/packages/tuxedo-drivers/-/issues/138 for a follow-up issue.