tuxedocomputers/tuxedo-keyboard #121: tuxedo-backlight-control does not start

xevilstar, 2022-03-03:
root@ghost:~# python -V
Python 3.9.10
root@ghost:~# /usr/share/tuxedo-backlight-control/ui.py
Traceback (most recent call last):
  File "/usr/share/tuxedo-backlight-control/ui.py", line 358, in <module>
    init()
  File "/usr/share/tuxedo-backlight-control/ui.py", line 353, in init
    App(root)
  File "/usr/share/tuxedo-backlight-control/ui.py", line 51, in __init__
    'color_left': tk.StringVar(self, value=backlight.color_left.capitalize()),
AttributeError: 'NoneType' object has no attribute 'capitalize'

vinzv, 2022-03-04:
You're messing up tuxedo-backlight-control with tuxedo-keyboard. The earlier it's not from us. For issues with tuxedo-backlight-control please ask here: https://github.com/webketje/tuxedo-backlight-control

closed 2022-03-04 by vinzv

xevilstar, 2022-03-04:
https://github.com/webketje/tuxedo-backlight-control/issues/20