tuxedocomputers/tuxedo-keyboard #122: No such thing as /sys/devices/platform/tuxedo_keyboard/ on my laptop

xevilstar, 2022-03-03:
backlight color red
Traceback (most recent call last):
  File "/usr/local/bin/backlight", line 69, in <module>
    backlight.state = 1
  File "/usr/share/tuxedo-backlight-control/backlight_control.py", line 122, in state
    self.set_device_param('state', str(value))
  File "/usr/share/tuxedo-backlight-control/backlight_control.py", line 70, in set_device_param
    filehandle = open(BacklightControl.DEVICE_PATH + prop, mode='w')
PermissionError: [Errno 13] Permission denied: '/sys/devices/platform/tuxedo_keyboard/state'
root@ghost:~# ls /sys/devices/platform/tuxedo_keyboard/state
ls: cannot access '/sys/devices/platform/tuxedo_keyboard/state': No such file or directory
root@ghost:~# lsmod|grep tuxedo
tuxedo_io              20480  0
tuxedo_keyboard        49152  3 tuxedo_io,uniwill_wmi,clevo_wmi
sparse_keymap          16384  2 asus_wmi,tuxedo_keyboard

vinzv, 2022-03-04:
See #121

closed 2022-03-04 by vinzv

xevilstar, 2022-03-04:
https://github.com/webketje/tuxedo-backlight-control/issues/19

xevilstar, 2022-03-04:
https://github.com/webketje/tuxedo-backlight-control/issues/20

webketje, 2022-03-05:
@vinzv this error will only occur if tuxedo_keyboard is not installed to SysFS at `/sys/devices/platform/tuxedo_keyboard`. So either 

- @xevilstar has mistaken tuxedo-backlight-control for a generic keyboard backlight utility, which it is not (please share your OS + version, what version of tuxedo-backlight-control & tuxedo_keyboard are installed)
- or tuxedo_keyboard no longer installs to SysFS? Was this perhaps changed recently?

vinzv, 2022-03-05:
As there were no recent code changes I suspect the first.

xevilstar, 2022-03-07:
recent debian bookworm tuxedo-backlight-control and tuxedo_keyboard downloaded from git

xevilstar, 2022-03-08:
Yes I have rebooted multiple times the leds work but I seem to be unable to change the color (missing state file)the keyboard chipset should be a clevo of some sort (I have made the producer make me a custom laptop at pcspecialist.it)
gcc --version
gcc (Debian 11.2.0-16) 11.2.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

ld --version
GNU ld (GNU Binutils for Debian) 2.38
Copyright (C) 2022 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.

since with the patched 5.16.x kernel the directory /sys/devices/platform/tuxedo_keyboard was totally missing I have installed the rc of the 5.17 kernel and the directory magically appeared but without the "state" filehttps://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git

backlight --version
0.8.0

since the ui doesn't start
backlight ui
Traceback (most recent call last):
  File "/usr/local/bin/backlight", line 43, in <module>
    init()
  File "/usr/share/tuxedo-backlight-control/ui.py", line 353, in init
    App(root)
  File "/usr/share/tuxedo-backlight-control/ui.py", line 51, in __init__
    'color_left': tk.StringVar(self, value=backlight.color_left.capitalize()),
AttributeError: 'NoneType' object has no attribute 'capitalize'

I am using the cli from bash/root account
cat /etc/apt/sources.list
# deb cdrom:[Debian GNU/Linux 11.2.0 _Bookworm_ - Unofficial amd64 NETINST with firmware 20211218-11:12]/ bookworm contrib main non-free

#deb cdrom:[Debian GNU/Linux 11.2.0 _Bookworm_ - Unofficial amd64 NETINST with firmware 20211218-11:12]/ bookworm contrib main non-free

deb http://deb.debian.org/debian/ bookworm main non-free contrib
deb-src http://deb.debian.org/debian/ bookworm main non-free contrib

deb http://security.debian.org/debian-security bookworm-security main contrib non-free
deb-src http://security.debian.org/debian-security bookworm-security main contrib non-free

# bookworm-updates, to get updates before a point release is made;
# see https://www.debian.org/doc/manuals/debian-reference/ch02.en.html#_updates_and_backports
deb http://deb.debian.org/debian/ bookworm-updates main contrib non-free
deb-src http://deb.debian.org/debian/ bookworm-updates main contrib non-free

# This system was installed using small removable media
# (e.g. netinst, live or single CD). The matching "deb cdrom"
# entries were disabled at the end of the installation process.
# For information about how to configure apt package sources,
# see the sources.list(5) manual.

uname -a
Linux ghost 5.17.0-rc7-amd64 #1 SMP PREEMPT Mon Mar 7 12:22:19 CET 2022 x86_64 GNU/Linux

lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 8087:0029 Intel Corp. AX200 Bluetooth
Bus 003 Device 005: ID 048d:6008 Integrated Technology Express, Inc. ITE Device(8291)
Bus 003 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 003 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 004: ID 045b:0210 Hitachi, Ltd 
Bus 002 Device 003: ID 05e3:0620 Genesys Logic, Inc. GL3523 Hub
Bus 002 Device 002: ID 0bda:0316 Realtek Semiconductor Corp. Card Reader
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 04f2:b71a Chicony Electronics Co., Ltd HD Webcam
Bus 001 Device 006: ID 046d:c07c Logitech, Inc. M-R0017 [G700s Rechargeable Gaming Mouse]
Bus 001 Device 005: ID 2a7a:8a47  CASUE USB KB
[... 832 lines cut ...]
       capabilities: platform
  *-input:6
       product: TUXEDO Keyboard
       physical id: 8
       logical name: input25
       logical name: /dev/input/event21
       capabilities: platform
  *-input:7
       product: Lid Switch
       physical id: 9
       logical name: input3
       logical name: /dev/input/event3
       capabilities: platform
  *-input:8
       product: Video Bus
       physical id: a
       logical name: input4
       logical name: /dev/input/event4
       capabilities: platform
  *-input:9
       product: Video Bus
       physical id: b
       logical name: input5
       logical name: /dev/input/event5
       capabilities: platform