tuxedocomputers/tuxedo-keyboard #124: Renaming My Project

salihmarangoz, 2022-03-06:
Hello,

I am Salih from Germany and I am currently developing https://github.com/salihmarangoz/system_tray_extensions . I would like to use the "tuxedo" keyword for renaming my repository to **"tuxedo-keyboard-effects"** also while indicating that the project is unofficial/fan-made. Even though this is a fan-made project I thought it would be better to ask before using a trademark keyword. Whatever your answer is the scope of this project will still remain same. I am waiting for an answer with yes or no.

Currently, I am re-implementing the project. I am also planning to fork TCC and apply modifications to add "Keyboard Effects" system tray menu which will be compatible with my project. Controlling RGB lights with modified TCC will be optional. Other than this, people will be able to control it using a separate tray app or with terminal commands. All of these are purely for educational purposes for me. 

While speaking about educational purposes, I also want people to be able to use their RGB keyboards with their custom Python scripts and make the interface super simple. So that people with little Python and Numpy knowledge will be able to control RGB lights and feel the joy of being able to control and customize their devices. I am planning to put some tutorials and simple examples too.

Note: I have sent a message using the website's contact page and I was asked to ask the devs directly.

Best,
Salih M.

vinzv, 2022-03-14:
Hi Salih,

thanks for getting in touch! And thanks for asking for permission, we appreciate that.

1. We talked about your request and hereby confirm that using "TUXEDO" as part of a tool's name is okay, as long as you state in a README that it's nothing official.

2. Regarding your tray icon changes and (even more there) forking TCC maybe it's better to avoid that and to add your changes "upstream"? This would give all users the chance to benefit from your great work. 
We are aware that this might not be an easy task, if even possible. So we're absolutely open to compensate your work in a way or the other.

Looking forward to hearing from you!

salihmarangoz, 2022-03-17:
Hello,

Thanks for confirming that. 

**Regarding the TCC:** I will try to implement a simple code for communicating over dbus (or zeromq) and allowing users to control the app through submenu. As another step it would be good to have a section near to "Dashboard", but I think this has to wait for a while. I will add this feature as disabled by default so it will be fine while distributing the modified TCC. If someone wants this feature they will be able to enable from "Global profile settings".

**Regarding my project:** I am still designing and trying different things. Currently, I am planning to combine ITE related userspace drivers (and maybe others used in Tuxedo devices) into a single python **Driver library**. Also, develop a separate **Client library** to give access individual projects use the framework which will communicate over DBUS (or zeromq). And lastly, **the core package** will include the Server, installation, its own virtual environment, effect files, etc. Depending on the complexity I am thinking about distributing the core package as a snap as an alternative to manual installation. I am thinking about ZeroMQ as an alternative method, so that people can implement bridges using it (e.g. Javascript, Cpp) and looks simpler compared to DBUS. And lastly, there will be a proper CLI interface (to allow people to combine it with their scripts) and a simple systemtray app for controlling the app (who doesn't use TCC and just wants to see some keyboard effects).

I am lacking in software design skills. If you have any recommendations or want to report a flaw about my thoughts/designs please correct me.

tuxedoxt, 2022-04-25:
Hello,

there was another brief internal discussion about the feature.

Regarding TCC integration. Provisional expectations on this feature as planned for development for TCC is that keyboard brightness, color, per-area/per-key be set to some kind of keyboard/lighting profile which can be assigned to a TCC profile. Implemented into existing code base.

Regarding keyboard backlight drivers, currently all are distributed as kernel modules (incl. ITE hid drivers). We're moving in the direction of standard `::kbd_backlight` led_classdev brightness interface. However I see no direct problem with user-space versions of the usb hid drivers since I believe this led interface can be created from user-space as well. Other backlight drivers must remain kernel modules however since communication through the EC is necessary.

Regarding your design, it sounds like a feasible project. Some parts could fit into TCC, others not, and some you explicitly want separate :) CLI interface sounds like a good idea for tinkerers. TCC gui to driver communication is currently done similar to what you describe (except for the cases of remaining config file changes) tcc-gui -> dbus -> tccd -> tuxedo-io or sysfs.

closed 2023-12-11 by tuxedo-bot