tuxedocomputers/tuxedo-keyboard #125: Expected timeline to get PR reviewed

brunoais, 2022-04-12:
What's the expected timeline to have PR reviewed?
[Mine](https://github.com/tuxedocomputers/tuxedo-keyboard/pull/105) has been sitting for about half an year with just me fixing when you made a breaking update.

jondo, 2023-02-26:
In the meantime the discussion within the affected PR continued. This should be closed (and asked within the PR in the first place).

brunoais, 2023-02-26:
@jondo Meh... I didn't get a direct answer but it's fine. I can close.
I'm still trying to keep up with the changes but being busy as I am, when I get an update done, there's another one to integrate into my changes. That takes away motivation, unfortunately.

closed 2023-02-26 by brunoais