tuxedocomputers/tuxedo-keyboard #127: kernel: tuxedo_keyboard: module verification failed: signature and/or required key missing - tainting kernel

qufo418, 2022-04-26:
Hi team,

**Problem (Current behavior):**
- the tuxedo-keyboard package is not signed, but it should. 

**Expected Behavior:**
- the tuxedo-keyboard package is signed and no such message (below) are shown in the system journal

**Seen on system:**
- Tuxedo Puls15 running Manjaro-Linux (kernel 5.17.1-3-MANJARO , DM Xfce)
- Installed versions:
```
tuxedo-keyboard                                3.0.9-4                extra 
    Keyboard Backlight Driver from TUXEDO
    Computers
tuxedo-control-center                          1.1.3-1                extra 
    A tool to help you control performance,
    energy, fan and comfort settings on
    TUXEDO laptops.
```
Error message taken from journalctl -xk:

```
kernel: tuxedo_keyboard: loading out-of-tree module taints kernel.
kernel: tuxedo_keyboard: module verification failed: signature and/or required key missing - tainting kernel
```

GiorgioSgl, 2022-05-20:
Same problem here on Fedora 36

GiorgioSgl, 2022-05-25:
I fixed it by disactivate the secure boot in the bios

qufo418, 2022-07-18:
> I fixed it by disactivate the secure boot in the bios

That's not really a fix, is it?

onurbbr, 2022-07-20:
U have to sign tuxedo_keyboard module. That's why ur module is not working.

Starcr0ss, 2023-05-06:
This issue still exists a year later. I have Secure Boot off.
```
[    1.972255] tuxedo_keyboard: loading out-of-tree module taints kernel.
[    1.972289] tuxedo_keyboard: module verification failed: signature and/or required key missing - tainting kernel
```
Kernel 6.3.1-arch1-1

Cyrille37, 2023-11-04:
Hi

Same with `Linux Mint 21.2 Cinnamon` with `Kernel 6.2.16-060216-generic`

From `dmesg` :
```
[    5.213618] tuxedo_keyboard: loading out-of-tree module taints kernel.
[    5.213722] tuxedo_keyboard: module verification failed: signature and/or required key missing - tainting kernel
[    5.215973] tuxedo_keyboard: module init
[    5.369271] tuxedo_keyboard: module init
```

closed 2023-12-11 by tuxedo-bot