tuxedocomputers/tuxedo-keyboard #130: Rust and upstreaming

AaronErhardt, 2022-06-30:
It feels a bit odd that TUXEDO has no upstream kernel drivers despite being a Linux hardware vendor. Especially on distros not officially supported by TUXEDO this requires some extra steps for users to setup the driver.

Given that Rust support will likely be added in version 5.20 of the kernel, the timing for upstreaming a new driver written in Rust would be perfect. Rust will also help with keeping the driver simpler, more readable and memory safe.

Since any activities regarding upstreaming this driver seem to have stalled I wondered what plans you have for the future and whether you'd be interested in a Rust version of this driver. Besides, I'd like to make the driver a bit smarter to reduce the sysfs overhead for fan controls and RGB animations (which is the main bottleneck in tuxedo-rs right now).

DeRaafMedia, 2022-07-21:
As a Tuxedo laptop user I would be more than happy if someone could make the driver available in standard kernel. It seems this project has died, and is slowly rotting away at the moment. 

Would love to have a working RGB back light.

ruzko, 2022-09-03:
Since there hasn't been any development here for a few months, and I can't write C or Rust worth worth a gram of salt, should we open a bug/feature bounty?

DeRaafMedia, 2022-09-03:
Would be nice to hear the thoughts of Tuxedo computers on it.

fethomm, 2022-11-02:
Upstreaming at the moment still waits on [#62](https://github.com/tuxedocomputers/tuxedo-keyboard/pull/62)

closed 2023-12-11 by tuxedo-bot