tuxedocomputers/tuxedo-keyboard #134: no temp no fan on 3.0.10

Qutn, 2022-07-19:
Hello,

On gentoo, the provided versions recently got bumped from 3.0.8 to 3.0.10 which induced some issues.

This is on a stellaris gen 3 15 intel.

In tuxedo control center, the temperature and fan speed display stay at 0
The special key to open the tuxedo control center does not work anymore

reverting back to 3.0.8 fixes the problems

reverting back to 3.0.9 does not.

in 3.0.10, the tccd service output this interesting info :
`tuxedo-io ver 0.2.4 [ interface: inactive ]`

In 3.0.8 the tccd service outputs this instead :
`tuxedo-io ver 0.2.4 [ interface: uniwill ]`

This was tested on multiple kernel versions : 5.18.12, 5.18.10 and 5.15 (lts)

on 3.0.10, dmesg tells me that tuxedo keyboard was loaded, nothing more.
on 3.0.8, dmesg first complains about me loading out of tree module, fair, then tells me this :
```
[    8.985492] tuxedo_keyboard: Model 'TUXEDO Stellaris Intel Gen3 (TGL)' found
[    8.985588] input: TUXEDO Keyboard as /devices/platform/tuxedo_keyboard/input/input16
```

thepasto, 2022-08-24:
Same here on Pulse 15 gen 1.
this happens when modrpobing the module, running on Archlinux -  5.19.3-arch1
```
[    6.941940] BPF: [112261] Invalid kind:19
[    6.941953] failed to validate module [tuxedo_keyboard] BTF: -22
[    7.045839] BPF: [112261] Invalid kind:19
[    7.045855] failed to validate module [tuxedo_keyboard] BTF: -22
```
Thanks

koutheir, 2022-08-24:
Same thing happening on ArchLinux with LTS kernel version `5.15.62-1-lts`.

thepasto, 2022-09-04:
Updating EC to v. 1.09.30 (Pulse 15 gen 1) seems to solve the issue

Qutn, 2022-09-05:
> Updating EC to v. 1.09.30 (Pulse 15 gen 1) seems to solve the issue

Hi, I'm sorry, what is EC ?

Qutn, 2022-09-08:
After being in contact with tuxedo, I saw that the bios and ec files updates were missing, I downloaded them and it seems that my bios was already up to date, nevertheless I tried updating again and it didn't fix anything, back to 3.0.8

I updated my Bios and EC to N.1.07A03 (BIOS) / 1.07.17 (EC)

tuxedoxt, 2022-10-17:
Hello,

are the problems persisting with the new versions? (currently TCC v1.2.1 + TK v3.1.0)

If log still outputs

> `tuxedo-io ver 0.3.0 [ interface: inactive ]`

run

```
sudo systemctl restart tccd
```

and check what the log says

Qutn, 2022-10-17:
Hello,

I updated tk to 3.1.0 and TCC to 1.2.1, unfortunately I still do not have the temperatures, nor the fan control and it seems that I now cannot reduce the cpu clock or number of cpus

this is the output of the logs from tccd with my last boot after update and after restarting tccd since the log was still outputting "interface: inactive"

```
Oct 17 16:29:36 [hostname] tccd[5610]: Daemon started
Oct 17 16:29:36 [hostname] tccd[5610]: Set display brightness to 70% (70) on nvidia_0
Oct 17 16:29:36 [hostname] tccd[5610]: tuxedo-io ver 0.3.0 [ interface: inactive ]
Oct 17 16:29:36 [hostname] tccd[5610]: Starting daemon v1.2.1 (node: v14.4.0 arch: x64)
Oct 17 16:29:36 [hostname] systemd[1]: Started TUXEDO Control Center Service.
Oct 17 16:29:36 [hostname] systemd[1]: tccd.service: Consumed 8.695s CPU time.
Oct 17 16:29:36 [hostname] systemd[1]: Stopped TUXEDO Control Center Service.
Oct 17 16:29:36 [hostname] systemd[1]: tccd.service: Deactivated successfully.
Oct 17 16:29:36 [hostname] tccd[5603]: Daemon is stopped
Oct 17 16:29:36 [hostname] tccd[4677]: DisplayBacklightWorker: Save display brightness 70% (70) on exit
Oct 17 16:29:36 [hostname] tccd[4677]: SIGINT - Exiting
Oct 17 16:29:36 [hostname] tccd[5603]: Stopping daemon..
Oct 17 16:29:36 [hostname] systemd[1]: Stopping TUXEDO Control Center Service...
Oct 17 16:20:40 [hostname] tccd[4677]: Daemon started
Oct 17 16:20:39 [hostname] tccd[4677]: Set display brightness to 70% (70) on nvidia_0
Oct 17 16:20:39 [hostname] tccd[4677]: tuxedo-io ver 0.3.0 [ interface: inactive ]
Oct 17 16:20:39 [hostname] tccd[4677]: Starting daemon v1.2.1 (node: v14.4.0 arch: x64)
Oct 17 16:20:39 [hostname] systemd[1]: Started TUXEDO Control Center Service.
Oct 17 16:20:39 [hostname] systemd[1]: tccd.service: Consumed 1.380s CPU time.
Oct 17 16:20:39 [hostname] systemd[1]: Stopped TUXEDO Control Center Service.
Oct 17 16:20:39 [hostname] systemd[1]: tccd.service: Deactivated successfully.
Oct 17 16:20:39 [hostname] tccd[4669]: Daemon is stopped
Oct 17 16:20:39 [hostname] tccd[4642]: DisplayBacklightWorker: Save display brightness 70% (70) on exit
Oct 17 16:20:39 [hostname] tccd[4642]: SIGINT - Exiting
Oct 17 16:20:39 [hostname] tccd[4669]: Stopping daemon..
Oct 17 16:20:39 [hostname] systemd[1]: Stopping TUXEDO Control Center Service...
Oct 17 16:20:09 [hostname] tccd[4642]: Daemon started
Oct 17 16:20:09 [hostname] tccd[4642]: Set display brightness to 70% (70) on nvidia_0
Oct 17 16:20:09 [hostname] tccd[4642]: tuxedo-io ver 0.3.0 [ interface: inactive ]
Oct 17 16:20:09 [hostname] tccd[4642]: Starting daemon v1.2.1 (node: v14.4.0 arch: x64)
Oct 17 16:20:09 [hostname] systemd[1]: Started TUXEDO Control Center Service.
Oct 17 16:20:09 [hostname] systemd[1]: tccd.service: Consumed 4.636s CPU time.
Oct 17 16:20:09 [hostname] systemd[1]: Stopped TUXEDO Control Center Service.
Oct 17 16:20:09 [hostname] systemd[1]: tccd.service: Deactivated successfully.
Oct 17 16:20:09 [hostname] tccd[4636]: Daemon is stopped
Oct 17 16:20:09 [hostname] tccd[2370]: DisplayBacklightWorker: Save display brightness 70% (70) on exit
Oct 17 16:20:09 [hostname] tccd[2370]: SIGINT - Exiting
Oct 17 16:20:09 [hostname] tccd[4636]: Stopping daemon..
Oct 17 16:20:09 [hostname] systemd[1]: Stopping TUXEDO Control Center Service...
Oct 17 16:15:20 [hostname] tccd[2370]: Daemon started
Oct 17 16:15:20 [hostname] tccd[2370]: Set display brightness to 70% (70) on nvidia_0
Oct 17 16:15:20 [hostname] tccd[2370]: Saved updated settings
Oct 17 16:15:20 [hostname] tccd[2370]: Missing state id assignment for 'BAT' but found profile by name 'min'
Oct 17 16:15:20 [hostname] tccd[2370]: Missing state id assignment for 'AC' but found profile by name 'med'
Oct 17 16:15:20 [hostname] tccd[2370]: Saved updated profiles
Oct 17 16:15:20 [hostname] tccd[2370]: (readProfiles) Generated id (yi4u23cszbl9cuzdko) for Summer Max
Oct 17 16:15:20 [hostname] tccd[2370]: (readProfiles) Generated id (2ebbs4x0uxel9cuzdko) for Summer Med
Oct 17 16:15:20 [hostname] tccd[2370]: (readProfiles) Generated id (45y27k3kjifl9cuzdko) for Lap
Oct 17 16:15:20 [hostname] tccd[2370]: (readProfiles) Generated id (3ytg78vazvwl9cuzdko) for max
Oct 17 16:15:20 [hostname] tccd[2370]: (readProfiles) Generated id (52i69s7uivl9cuzdko) for med
Oct 17 16:15:20 [hostname] tccd[2370]: (readProfiles) Generated id (1m1zaincppyl9cuzdko) for min
Oct 17 16:15:20 [hostname] tccd[2370]: (readProfiles) Generated id (bbazlzrjxrbl9cuzdkn) for Default custom profile
Oct 17 16:15:20 [hostname] tccd[2370]: tuxedo-io ver 0.3.0 [ interface: inactive ]
Oct 17 16:15:20 [hostname] tccd[2370]: Starting daemon v1.2.1 (node: v14.4.0 arch: x64)
Oct 17 16:15:19 [hostname] systemd[1]: Started TUXEDO Control Center Service.
```

also it may be interesting but that is what my kernel log say when I grep for tuxedo

```
Oct 17 16:15:15 [hostname] kernel: tuxedo_keyboard: loading out-of-tree module taints kernel.
Oct 17 16:15:15 [hostname] kernel: tuxedo_keyboard: module init
Oct 17 16:15:15 [hostname] kernel: tuxedo_keyboard: no active interface while read addr 0x0740
Oct 17 16:15:15 [hostname] kernel: acpi PNP0C14:01: duplicate WMI GUID 05901221-D566-11D1-B2F0-00A0C9062910 (first instance was on PNP0C14:00)
Oct 17 16:15:20 [hostname] kernel: tuxedo_keyboard: no active interface while read addr 0x0740
Oct 17 16:15:20 [hostname] kernel: tuxedo_keyboard: no active interface while read addr 0x0740
Oct 17 16:15:20 [hostname] kernel: tuxedo_keyboard: no active interface while read addr 0x0740
[this goes on forever]
```

I feel like this may be due to how those two are packaged under gentoo, I will try to look into it

Qutn, 2023-03-09:
Updating to tuxedo-control-center-1.2.5 and tuxedo-keyboard to 3.1.4 fixed this issue

closed 2023-03-09 by Qutn