tuxedocomputers/tuxedo-keyboard #138: package for opensuse leap 15.4 doesn't works after a normal software update

procuste, 2022-09-17:
as in the title I use this repo:
https://rpm.tuxedocomputers.com/opensuse/15.4/
in my leap 15.4 but, after any normal software update, backlit stop to work, I have to unconditionally reinstall the package to have backlit working, as discussed in this thread:
https://forums.opensuse.org/showthread.php/575557-is-there-a-way-to-unconditionally-install-a-package?p=3159436#post3159436
could you please solve this problem??
manythanks, ciao, pier

tuxedoxt, 2022-11-08:
Hello,

little update, coincidentally noticed by some suse vanilla installations leap/tumbleweed that the systemd dkms service is not enabled by default.

try

```
systemctl enable dkms
```

and reboot should you run into the issue on suse.

Apflkuacha, 2023-01-20:
Another problem occured after updating the tuxedo-keyboard from 3.1.2 to 3.1.3: in dkms there is following error message
`Error! Could not locate dkms.conf file. File: /var/lib/dkms/tuxedo-keyboard/3.1.2/source/dkms.conf does not exist.`

Reason: the files in `/var/lib/dkms/tuxedo-keyboard/3.1.2/` were not deleted but the ones here `/usr/src/tuxedo-keyboard-3.1.2` are gone.

tuxedoxt, 2023-01-20:
> 

Hmm weird. Using the rpm package? I'd be interested in logs from the package upgrade.

Apflkuacha, 2023-01-20:
Here you could see the logfile entries from /var/log/zypp/history 
Updates were made using the Discover Software Center on my Tumbleweed KDE.

<details>
<summary>Click this to see the log files of the installation of tuxedo-keyboard 3.1.3</summary>

```
2023-01-12 15:24:28|remove |libsrt1_4|1.4.4-1.4|x86_64||
# 2023-01-12 15:24:51 tuxedo-keyboard-3.1.3-1.noarch.rpm installed ok
# Additional rpm output:
# WARNING: /usr/lib/dkms/common.postinst does not exist.
# Loading new tuxedo-keyboard-3.1.3 DKMS files...
# Building for 6.1.3-1-default
# Building initial module for 6.1.3-1-default
# Done.
# 
# tuxedo_keyboard.ko.zst:
# Running module version sanity check.
#  - Original module
#    - This kernel never originally had a module by this name
#  - Installation
#    - Installing to /lib/modules/6.1.3-1-default/kernel/lib//
# 
# clevo_wmi.ko.zst:
# Running module version sanity check.
# 
# Good news! Module version 0.0.4 for clevo_wmi.ko.zst
# exactly matches what is already found in kernel 6.1.3-1-default.
# DKMS will not replace this module.
# You may override by specifying --force.
# 
# clevo_acpi.ko.zst:
# Running module version sanity check.
# 
# Good news! Module version 0.0.3 for clevo_acpi.ko.zst
# exactly matches what is already found in kernel 6.1.3-1-default.
# DKMS will not replace this module.
# You may override by specifying --force.
# 
# tuxedo_io.ko.zst:
# Running module version sanity check.
#  - Original module
#    - This kernel never originally had a module by this name
#  - Installation
#    - Installing to /lib/modules/6.1.3-1-default/kernel/lib//
# 
# uniwill_wmi.ko.zst:
# Running module version sanity check.
# 
# Good news! Module version 0.0.1 for uniwill_wmi.ko.zst
# exactly matches what is already found in kernel 6.1.3-1-default.
# DKMS will not replace this module.
# You may override by specifying --force.
# Adding any weak-modules
# 
# depmod....
# 
# DKMS: install completed.
# Check tccd running status
# Stop tccd temporarily
# /var/tmp/rpm-tmp.Z0111s: line 21: fg: no job control
# (Re)load modules if possible
# Start tccd again
# 
# Uninstall of tuxedo-keyboard module (version 3.1.2-1) beginning:
# 
# -------- Uninstall Beginning --------
# Module:  tuxedo-keyboard
# Version: 3.1.2
# Kernel:  6.1.2-1-default (x86_64)
[... 111 lines cut ...]
# uniwill_wmi.ko.zst:
# Running module version sanity check.
#  - Original module
#    - No original module exists within this kernel
#  - Installation
#    - Installing to /lib/modules/6.0.12-1-default/kernel/lib//
# Adding any weak-modules
# 
# depmod....
# 
# DKMS: install completed.
# Check tccd running status
# Stop tccd temporarily
# /var/tmp/rpm-tmp.2XAqac: line 21: fg: no job control
# (Re)load modules if possible
# Start tccd again
# 
# Uninstall of tuxedo-keyboard module (version 3.1.1-1) beginning:
# Error! The module/version combo: tuxedo-keyboard-3.1.1
# is not located in the DKMS tree.
# 
2022-12-21 08:41:36|install|tuxedo-keyboard|3.1.2-1|noarch||Tuxedo_15.4|f561a01d52acf4a5516fcf0be25a6a5147db092f24088608ab9312c37a3b2337|
```

</details>

alexbradd, 2023-07-04:
Can confirm that this problem is still present when upgrading up to version 3.2.7 (latest release at the time of writing) on tumbleweed with latest dkms.

RPM scriptlet output:

```
#  - Original module
#    - No original module was found for this module on this kernel.
#    - Use the dkms install command to reinstall any previous module version.
# 
# clevo_acpi.ko.zst:
#  - Uninstallation
#    - Deleting from: /lib/modules/6.3.7-1-default/updates/
#  - Original module
#    - No original module was found for this module on this kernel.
#    - Use the dkms install command to reinstall any previous module version.
# 
# tuxedo_io.ko.zst:
#  - Uninstallation
#    - Deleting from: /lib/modules/6.3.7-1-default/updates/
#  - Original module
#    - No original module was found for this module on this kernel.
#    - Use the dkms install command to reinstall any previous module version.
# 
# uniwill_wmi.ko.zst:
#  - Uninstallation
#    - Deleting from: /lib/modules/6.3.7-1-default/updates/
#  - Original module
#    - No original module was found for this module on this kernel.
#    - Use the dkms install command to reinstall any previous module version.
# depmod....
# 
```

I did some very light debugging and it seems that after upgrade dkms is left in an inconsistent state  thinking that the previous version is still installed and thus it is searching for the old version's `dkms.conf` which has been removed by RPM. This makes it impossible to do anything other than forcefully cleaning the RPM tree and re-adding everything by hand.

I might do some more digging after I am finished with exams this summer and let you know.

closed 2023-12-11 by tuxedo-bot