tuxedocomputers/tuxedo-keyboard #144: Not working on Stellaris 15 with Debian Testing

Stummi, 2022-12-22:
I just got a Stellaris 15, and installed a fresh debian Testing. I followed all the instruction but my backlight doesn't work.

The module got successfully loaded

```
~$ lsmod | grep tuxedo
tuxedo_io              24576  0
tuxedo_keyboard        57344  3 tuxedo_io,uniwill_wmi,clevo_wmi

~$ sudo dmesg | grep tuxedo
[....]
[   19.314946] tuxedo_keyboard: module init
[   19.315825] tuxedo_keyboard: no active interface while read addr 0x0740
[   19.315827] tuxedo_keyboard: no active interface while read addr 0x0742
[   19.315828] tuxedo_keyboard: no active interface while read addr 0x078e
[   21.485070] input: TUXEDO Keyboard as /devices/platform/tuxedo_keyboard/input/input38
```

What I notice first is that the documented path sys exists but it has none of the documented files in it
```
~$ ls /sys/devices/platform/tuxedo_keyboard/ 
charging_profile  driver  driver_override  input  leds  modalias  power  subsystem  uevent
~$ ls /sys/devices/platform/tuxedo_keyboard/input/
input38
~$ ls /sys/devices/platform/tuxedo_keyboard/input/input38/
capabilities  device  event260  id  inhibited  modalias  name  phys  power  properties  subsystem  uevent  uniq
~$ ls /sys/devices/platform/tuxedo_keyboard/leds/
lightbar_animation::status  lightbar_rgb:1:status  lightbar_rgb:2:status  lightbar_rgb:3:status
```

However they are to be found on another path (and only there): 
```
~$ find /sys | grep tuxedo | grep mode
/sys/module/tuxedo_keyboard/parameters/mode
```

Writing into these files is not possible though

```
root@host:/sys/module/tuxedo_keyboard/parameters# echo 1 > mode
bash: echo: write error: Input/output error
```

So, any hints what I could be doing wrong here?

Qutn, 2023-03-09:
not much of a help, but I have the same issue on gentoo with the latest tuxedo_keyboard and tuxedo-control-center. However I can manipulate the lightbar at /sys/devices/platform/tuxedo_keyboard/leds

patrickbuchner, 2023-04-07:
I also have a Stellaris AMD 15 Gen 3, which does not show any of the expected files in `/sys/devices/platform/tuxedo_keyboard/leds/`.
I am on Fedora 37 and have installed the version of `tuxedo-keyboard       noarch 3.2.0-1         copr:copr.fedorainfracloud.org:kallepm:tuxedo-keyboard`.
I have also made a fresh install with OpenSuse Leap 15.4. There I have installed the necessary repos as [described](https://www.tuxedocomputers.com/de/Infos/Hilfe-und-Support/Anleitungen/TUXEDO-Software-Paketquellen-hinzufuegen.tuxedo), after which I found the same problem.

tuxedoxt, 2023-04-11:
Hello,

Stellaris (all Gen:s) keyboard backlight still needs package `tuxedo-keyboard-ite`.

patrickbuchner, 2023-04-11:
Thank you, now the leds are showing up and with the rpm packages from https://rpm.tuxedocomputers.com/opensuse/15.4/ it works also for fedora
- [tuxedo-control-center_2.0.2.rpm](https://rpm.tuxedocomputers.com/opensuse/15.4/x86_64/tuxedo-control-center_2.0.2.rpm)
- [tuxedo-keyboard-3.2.1-1.noarch.rpm](https://rpm.tuxedocomputers.com/opensuse/15.4/noarch/tuxedo-keyboard-3.2.1-1.noarch.rpm)
- [tuxedo-keyboard-ite-0.3.0-1.noarch.rpm](https://rpm.tuxedocomputers.com/opensuse/15.4/noarch/tuxedo-keyboard-ite-0.3.0-1.noarch.rpm)

closed 2023-12-11 by tuxedo-bot