tuxedocomputers/tuxedo-keyboard #148: No back light on  InfinityBook Pro Gen7 (MK1) keyboard

Jeansen, 2022-12-26:
```
      _,met$$$$$gg.          
    ,g$$$$$$$$$$$$$$$P.       ------------------ 
  ,g$$P"     """Y$$.".        OS: Debian GNU/Linux bookworm/sid x86_64 
 ,$$P'              `$$$.     Model: TUXEDO InfinityBook Pro Gen7 (MK1) Standard 
',$$P       ,ggs.     `$$b:   Kernel: 6.0.0-6-amd64 
`d$$'     ,$P"'   .    $$$    Uptime: 17 mins 
 $$P      d$'     ,    $$P    Packages: 8428 (dpkg), 9 (flatpak), 7 (snap) 
 $$:      $$.   -    ,d$$'    Shell: bash 5.2.2 
 $$;      Y$b._   _,d$P'      Resolution: 2560x1600 
 Y$$.    `.`"Y$$$$P"'         DE: GNOME 43.1 
 `$$b      "-.__              WM: Mutter 
  `Y$$                        WM Theme: Adwaita 
   `Y$$.                      Theme: Materia-dark-compact [GTK2/3] 
     `$$b.                    Icons: Paper [GTK2/3] 
       `Y$$b.                 Terminal: guake 
          `"Y$b._             CPU: 12th Gen Intel i7-12700H (20) @ 4.600GHz 
              `"""            GPU: Intel Alder Lake-P 
                              Memory: 15836MiB / 64050MiB 

```

After running the DKMS make script, the modprobe install just wen fine. I've build and installed TCC and that works fine, too. 

I simply cannot get the back light for my keyboard to glow up ....

StevenSeifried, 2022-12-26:
`Fn + Space` toggles the keyboard backlight

Jeansen, 2022-12-27:
@StevenSeifried It works! But where do I find THAT information? (or what did I miss?)

closed 2022-12-27 by Jeansen