tuxedocomputers/tuxedo-keyboard #156: Fn F11 (airplane mode) not working

squalou, 2023-01-20:
Hi,

I have a TUXEDO InfinityBook Pro 14 Gen6 Standard,
I installed tuxedo-control-center, tuxedo-keyboard, tuxedo-touchpad-switch

I'm running PopOS at the moment.
I have the same issue under Wayland or X.

`xev` does not report any input for Fn F11. (for other Fn keys it does and it works fine. F11 alone works)

The Fn F11 key is supposed to trigger Airplane Mode but does nothing.

It's not a big deal, I mapped my own script to Super-F11 key, and it works fine. (I'm used to this, my previous xps13 had the same issues)

But I'd like to know if I'm missing an obvious thing.

Thank you !

regards

bortzmeyer, 2023-08-28:
Same problem with my Pulse 14. The `setkeycodes e078 666` does not seem to produce any effect.

develroo, 2023-10-24:
Yeah. same here on a Gen 1 Pulse 15 with Debian Testing 6.5.0-2-amd64

```
atkbd serio0: Unknown key released(translated set 2, code 0xf8 on isa0060/serio0).
atkbd serio0: Use 'setkeycodes e078<keycode>' to make it known.  
```

All the function key functions work in Gnome, so touchpad disable, aeroplane mode, keyboard backlight etc.

So it is working ok, just have this annoying dmesg warning every time the `Fn` is pressed. 

Kinda annoying that, as I hate spams in my logs.

closed 2023-12-11 by tuxedo-bot