tuxedocomputers/tuxedo-keyboard #160: Irregular input latency on Arch/Endeavour OS

vmaurin, 2023-01-28:
Hi

When using the module, I have a visible input lag on the keyboard, that is also quite irregular (sometime short, sometime long). It is not a system issue (trackpad is reponsive, external USB keyboard too).
If I remove the tuxedo_keyboard module, it works fine too, but I cannot have the control center anymore

A small video to expose the problem (where you can hear the key press) 

https://user-images.githubusercontent.com/17569830/215269422-385b6774-0794-4f90-b463-276e2ff45695.mp4

Laptop : Pulse 15
Kernel version : 6.1.8

crisidev, 2023-01-30:
I have the same issue on a pulse 14 with Ubuntu 22.04.

vmaurin, 2023-02-01:
@crisidev Which kernel ? From the source code, I am not seeing something handling a input loop or similar, so it might be an issue that show up with a specific kernel and some options on the module ?

crisidev, 2023-02-01:
It's something I experience from when I have the laptop, so various 5.x, 6.0, 6.1.

vmaurin, 2023-02-01:
Doing some test with "evtest", I can see two different sources in /dev/input/event[N]
* one `AT Translated Set 2 keyboard` that gets events when typing almost all the keys
* one `TUXEDO Keyboard`

It seems the latency affect both, but it might be a difference compared to a USB keyboard

vmaurin, 2023-02-02:
I have played with the code and recompile the module but no luck in this direction, but then I noticed something : 
when I stop the service tccd (the control center associated service), the lag disappears. It is showing up again as soon as I start the service

@crisidev could you try ?

```
sudo systemctl stop tccd
```

vmaurin, 2023-02-02:
Ok it was known already here https://github.com/tuxedocomputers/tuxedo-control-center/issues/262

vmaurin, 2023-02-02:
Closing this one as waiting on a fix on tccd