tuxedocomputers/tuxedo-keyboard #163: Modules fail to load on kernel 6.2.1

devsaurus, 2023-03-06:
I run SUSE Tumbleweed on my Pulse 14 (TUXEDO Pulse 14 Gen1/PULSE1401, BIOS N.1.07.A05 04/25/2022) and stumbled upon an issue with the modules provided by tuxedo-keyboard 3.1.4.

After upgrading to kernel 6.2.1-1-default, all tuxedo-keyboard modules fail to load with the same error:
```
<root shell>
# modprobe tuxedo_keyboard 
modprobe: ERROR: could not insert 'tuxedo_keyboard': Operation not permitted
# modprobe tuxedo_io 
modprobe: ERROR: could not insert 'tuxedo_io': Operation not permitted
# modprobe clevo_acpi 
modprobe: ERROR: could not insert 'clevo_acpi': Operation not permitted
# modprobe clevo_wmi 
modprobe: ERROR: could not insert 'clevo_wmi': Operation not permitted
# modprobe uniwill_wmi 
modprobe: ERROR: could not insert 'uniwill_wmi': Operation not permitted
```
As a result, fan control etc isn't effective and Tuxedo Control Center can't set any profiles.

Modules work fine on 6.2.0-1-default and load automatically during system startup.

What could prevent the modules from loading in 6.2.1 while they work in 6.2.0?

DeRaafMedia, 2023-03-09:
Same on Pop!_OS 

modprobe: FATAL: Module tuxedo_keyboard not found in directory /lib/modules/6.2.0-76060200-generic
modprobe: FATAL: Module tuxedo_io not found in directory /lib/modules/6.2.0-76060200-generic
modprobe: FATAL: Module clevo_acpi not found in directory /lib/modules/6.2.0-76060200-generic
modprobe: FATAL: Module clevo_wmi not found in directory /lib/modules/6.2.0-76060200-generic
modprobe: FATAL: Module uniwill_wmi not found in directory /lib/modules/6.2.0-76060200-generic

But now I know for sure the keyboard backlight is capable to produce different colors.

DeRaafMedia, 2023-03-09:
I did a reinstall of tuxedo-control-center via the terminal methode described on [here](https://www.tuxedocomputers.com/en/Infos/Help-and-Support/Instructions/Add-TUXEDO-Computers-software-package-sources.tuxedo#). 

modules are being found with `modprobe clevo_acpi` operation not permitted:
`modprobe: ERROR: could not insert 'clevo_acpi': Operation not permitted`

I think every works. But the LED on the dedicated Tuxedo Control Center button is lighting up constant, so probably there are other things not quitte right. My space bar for instance seem to be registering key presses really, really slow.

lluises, 2023-03-09:
Same problem here.

Failed to load `tuxedo_keyboard` on `6.2.1-arch1-1`, loosing control on the keyboard RGB once the system is up.

tuxedoxt, 2023-03-10:
Hello,

weird, build and inserted fine on a fresh tumbleweed install.

Maybe other special settings like.. secureboot or something not allowing unsigned modules?

lluises, 2023-03-10:
I fixed my issue.

It turned out that I was running an outdated version of the code. I update it and everything is working as expected.

Sorry for the inconvenience :v

devsaurus, 2023-03-10:
> Maybe other special settings like.. secureboot or something not allowing unsigned modules?

Good point, my laptop has secureboot enabled.

Just gave it a try and disabled the secureboot option in BIOS. The tuxedo-keyboard modules load without issue now on 6.2.1.

| Secureboot | Kernel | Module state |
| ----------------- | --------- | --------------------|
| enabled       | 6.2.0    | loads ok         |
| enabled       | 6.2.1    | doesn't load  |
| disabled      | 6.2.0    | loads ok         |
| disabled      | 6.2.1     | loads ok        |

Seems to be related to signing.

closed 2023-03-10 by devsaurus