tuxedocomputers/tuxedo-keyboard #167: Kernel 6.2.6 can't install from repo

jmaargh, 2023-03-30:
I'm running PopOS 22.04 on Tuxedo hardware and have some software installed from the Tuxedo repo.

The latest update broke with kernel 6.2.6 as the `tuxedo-keyboard` dkms module won't build. Relevant output from `apt upgrade`:

```
depmod...
Building initial module for 6.2.6-76060206-generic
ERROR (dkms apport): kernel package linux-headers-6.2.6-76060206-generic is not supported
Error! Bad return status for module build on kernel: 6.2.6-76060206-generic (amd64)
Consult /var/lib/dkms/tuxedo-keyboard/3.2.1/build/make.log for more information.
dpkg: error processing package tuxedo-keyboard (--configure):
 installed tuxedo-keyboard package post-installation script subprocess returned error exit status 10
dpkg: dependency problems prevent configuration of tuxedo-control-center:
 tuxedo-control-center depends on tuxedo-keyboard (>= 3.1.2); however:
  Package tuxedo-keyboard is not configured yet.
```

Any word on when this might be updated?

Other relevant details:

```
 $ apt policy tuxedo-keyboard
tuxedo-keyboard:
  Installed: 3.2.1
  Candidate: 3.2.1
  Version table:
 *** 3.2.1 500
        500 https://deb.tuxedocomputers.com/ubuntu jammy/main amd64 Packages
        100 /var/lib/dpkg/status
```

```
 $ modinfo tuxedo_keyboard
filename:       /lib/modules/6.1.11-76060111-generic/kernel/lib/tuxedo_keyboard.ko
version:        3.2.1
license:        GPL
description:    TUXEDO Computers keyboard & keyboard backlight Driver
author:         TUXEDO Computers GmbH <tux@tuxedocomputers.com>
license:        GPL
license:        GPL
srcversion:     C636E85C28026D987A2B09E
depends:        sparse-keymap,led-class-multicolor
retpoline:      Y
name:           tuxedo_keyboard
vermagic:       6.1.11-76060111-generic SMP preempt mod_unload modversions 
parm:           kbd_backlight_mode:Set the keyboard backlight mode
```

jmaargh, 2023-04-02:
Update:

I was wrong that the kernel version mismatch was the issue. The build log simply stated I was missing the required `gcc-12`. Surely this should have been handled by dependencies though?

rowie, 2023-05-07:
Installing gcc-12 fixes the issue for me! thx a lot man!

closed 2023-06-06 by vinzv