tuxedocomputers/tuxedo-keyboard #168: XMG APEX (Mid 2021) - lost keyboard backlight control aftert upgrade to tuxedo-keyboard 3.2.0 (and 3.2.1 too)

rurus9, 2023-03-31:
After upgrade tuxedo-keyboard to 3.2.0 and 3.2.1 (also necessarily TCC to 2.0.0, 2.0.1 and and recently 2.0.2)  my keyboard backlight is out of my control.

Before tuxedo-keyboard 3.2.0 and  TCC 2.0 and  I have configured tuxedo_keyboard module in this way:
```
$ cat /etc/modprobe.d/tuxedo_keyboard.conf
options tuxedo-keyboard mode=0 brightness=75 color_left=0xFFFFFF color_center=0xFFFFFF color_right=0xFFFFFF
```
it worked as intended, after starting the laptop, the BIOS set the backlight to a strong blue, after loading the system it changed according to my settings.

After upgrade to tuxedo-keyboard 3.2.0 backlight stopped worked (the keyboard backlight turns off when the system is loaded).
dmesg showed:
```
Mar 26 22:23:19 sam kernel: [  416.591995] tuxedo_keyboard: unknown parameter 'mode' ignored
Mar 26 22:23:19 sam kernel: [  416.591998] tuxedo_keyboard: unknown parameter 'brightness' ignored
Mar 26 22:23:19 sam kernel: [  416.591999] tuxedo_keyboard: unknown parameter 'color_left' ignored
Mar 26 22:23:19 sam kernel: [  416.592000] tuxedo_keyboard: unknown parameter 'color_center' ignored
Mar 26 22:23:19 sam kernel: [  416.592000] tuxedo_keyboard: unknown parameter 'color_right' ignored
```
So I changed /etc/modprobe.d/tuxedo_keyboard.conf to:
```
options tuxedo-keyboard kbd_backlight_mode=0
```
Now dmesg shows:
```
$ sudo dmesg | grep tuxedo | grep -i keybo
[   11.149945] tuxedo_keyboard: module init
[   11.175962] tuxedo_keyboard: Set keyboard backlight mode on CUSTOM
[   11.180676] tuxedo_keyboard: Set keyboard enabled to: 1
[   11.183150] input: TUXEDO Keyboard as /devices/platform/tuxedo_keyboard/input/input18
[   78.480919] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 1, new_color_code FF0000
[   78.907833] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 2, new_color_code FF00
[   79.479675] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 3, new_color_code FF
[   80.527184] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 4, new_color_code FFFF00
[   80.747057] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 5, new_color_code FF00FF
[   84.051120] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 6, new_color_code FFFF
[   84.366938] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 7, new_color_code FFFFFF
[   85.614452] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 1, new_color_code FF0000
[   85.855084] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 2, new_color_code FF00
[   86.087130] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 3, new_color_code FF
[   94.594886] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 4, new_color_code FFFF00
[   94.974746] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 5, new_color_code FF00FF
[   95.325048] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 6, new_color_code FFFF
[   95.710855] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 7, new_color_code FFFFFF
[  112.666727] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 1, new_color_code FF0000
[  113.562735] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 2, new_color_code FF00
[  115.802843] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 3, new_color_code FF
[  116.730724] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 4, new_color_code FFFF00
[  117.698662] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 5, new_color_code FF00FF
[  121.046768] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 6, new_color_code FFFF
[  122.058727] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 7, new_color_code FFFFFF
[  124.386817] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 1, new_color_code FF0000
[  125.798989] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 2, new_color_code FF00
[  126.413527] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 3, new_color_code FF
[  126.838752] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 4, new_color_code FFFF00
[  127.698518] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 5, new_color_code FF00FF
[  128.006765] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 6, new_color_code FFFF
```
but... my keyboard backlight is off (shuts down when the system loads), once it did not turn off, it lit up by default as set by BIOS.

Shortcut for change backlight also stopped working (it seems to me that it was [Fn] + [space], I have not used it for a long time, but it looks similar to: https://www.reddit.com/r/tuxedocomputers/comments/125z5gu/pusle_15_gen_1_keyboard_backlight_issues_circa/).

TCC shows:
"Your device has no keyboard backlight or is not (yet) supported by this control interface"
(but the fan control works as in previous versions).

tuxed_control and TCC install from the official repository: https://deb.tuxedocomputers.com/ubuntu

```
$ uname -a
Linux sam 6.2.0-10005-tuxedo #5 SMP PREEMPT_DYNAMIC Wed Mar 22 12:42:40 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

$ lsmod | grep tuxedo
tuxedo_io              24576  0
tuxedo_keyboard        73728  3 clevo_acpi,tuxedo_io,clevo_wmi
led_class_multicolor    16384  1 tuxedo_keyboard
sparse_keymap          16384  1 tuxedo_keyboard

$ ls /sys/class/leds/
input17::capslock  input17::kana     input17::scrolllock  input3::numlock     phy0-led
input17::compose   input17::numlock  input3::capslock     input3::scrolllock

$ ls /sys/devices/platform/tuxedo_keyboard/
driver  driver_override  input  modalias  power  subsystem  uevent
```

Matombo, 2023-03-31:
If I'm not mistaken that device has it's backlight controll keys on FN+Keypad+-*/

But the keyboard backlight should show up in /sys/class/leds/ which it doesn't.

Can you give me the output of `dkms status` and `sudo dmesg`?

MrHappy, 2023-07-30:
Is this resolved by someone? I ran into this too. Only thing that does work is changing the color (FN+Keypad /).

closed 2023-12-11 by tuxedo-bot