tuxedocomputers/tuxedo-keyboard #169: [Fn] + [space] backlight shortcut doesn't work anymore on Polaris 17 after TCC update

Uhter, 2023-04-02:
Hello,

After update to Tuxedo Control Center v2.0.2, Keyboard Backlight tool appeared and work well but **[Fn] + [space] backlight shortcut doesn't work anymore**: nothing happens when I use it :(

Configuration:

- TUXEDO Polaris 17 AMD Gen3
- BIOS: N.1.50A12
- EC: 1.32.29
- Manjaro Xfce
- kernel 6.2.8-1 and headers 6.2.8-1
- TTC 2.0.2-1
- Tuxedo keyboard dkms 3.2.1-1

what can I do?

tuxedoxt, 2023-04-03:
Hello,

first, make sure you reboot once after upgrade and try again.

Otherwise. Fn+backlight key(s) now trigger the usually used key events upon which the WM usually reacts. Last I checked Xfce does not implement all of these. I don't know the current state.

Uhter, 2023-04-03:
Hello @tuxedoxt,

So, this is not a Tuxedo/TCC bugs?

If I understand correctly, I have to assign a command to a Xfce shortcut? 
I know how to do, but what is this command/script?

Uhter, 2023-05-13:
Thanks @tuxedoxt but I failed to fix it this way. But… 

[Fn] + [space] works again since I installed kernel 6.3.0-1 and upgraded Tuxedo Control Center to 2.0.5 (but I can't change keyboard color because "_Your device has no keyboard backlight or is not (yet) supported by this control interface._").

I give up.
I think Polaris 17 + Manjaro don't make a good marriage.

closed 2023-12-11 by tuxedo-bot