tuxedocomputers/tuxedo-keyboard #173: Backlight on Suse Tumbleweed is not working

schrapm3, 2023-04-08:
Hi there,

the backlight on current Tumbleweed is not working although the module seems to be loading:

[    5.361621] tuxedo_keyboard: loading out-of-tree module taints kernel.
[    5.361658] tuxedo_keyboard: module verification failed: signature and/or required key missing - tainting kernel
[    5.362464] tuxedo_keyboard: module init
[    7.215943] input: TUXEDO Keyboard as /devices/platform/tuxedo_keyboard/input/input33

Any ideas ?

temirg, 2023-04-09:
@schrapm3: please check following:
TCC dashboard => Tools => Keyboard backlight (or similar in English) => brightness: 0% ?

Regards, temirg.

schrapm3, 2023-04-09:
Hi,

tried with Arch Linux on another SSD with the same result. Dmesg shows:

[    1.370437] tuxedo_keyboard: loading out-of-tree module taints kernel.
[    1.370487] tuxedo_keyboard: module verification failed: signature and/or required key missing - tainting kernel
[    1.371454] tuxedo_keyboard: module init
[    2.560862] input: TUXEDO Keyboard as /devices/platform/tuxedo_keyboard/input/input10

In TCC it says:

[image]

temirg, 2023-04-09:
Possible the reinstall would help:
zypper in -f tuxedo-keyboard tuxedo-control-center
/usr/libexec/dkms/dkms_autoinstaller start $(uname -r)
reboot

temirg, 2023-04-10:
A tuxedo device?

tuxedoxt, 2023-04-11:
Hello,

which device do you have?

groenator, 2023-07-24:
Hi, 

I am using Stellaris 16 gen 5 and I have the same issue. The backlight is not available on TW. 

The control center shows this:
[image]

Installed version:
[image]

I run the command above and I got the following error:

```
user@tuxedo:~> sudo zypper in -f tuxedo-keyboard tuxedo-control-center /usr/libexec/dkms/dkms_autoinstaller start $(uname -r)
Loading repository data...
Reading installed packages...
Forcing installation of 'tuxedo-keyboard-3.2.8-1.noarch' from repository 'TUXEDO'.
Forcing installation of 'tuxedo-control-center-2.0.8-1.x86_64' from repository 'TUXEDO'.
'6.4.3-1-default' not found in package names. Trying capabilities.
No provider of '6.4.3-1-default' found.
'/usr/libexec/dkms/dkms_autoinstaller' not found in package names. Trying capabilities.
No provider of '/usr/libexec/dkms/dkms_autoinstaller' found.
'start' not found in package names. Trying capabilities.
No provider of 'start' found.
Resolving package dependencies...

The following 2 packages are going to be reinstalled:
  tuxedo-control-center tuxedo-keyboard

groenator, 2023-07-24:
Installing the tuxedo-keyboard-ite package worked. The backlight is working on TW.

closed 2023-07-30 by vinzv