tuxedocomputers/tuxedo-keyboard #174: Keyboard backlight puts itself on !

nicobo, 2023-04-14:
Since several weeks (month?) the backlight of my InfinityBook S 15 keeps putting itself on alone from time to time.
It never did before, It was staying off until I put it on by the dedicated keys.
I've no idea why nor if it's a conflict with another key combination or a deamon that starts itself but it looks like it's like this since a tuxedo update.

nicobo, 2023-04-19:
> Hello,
> 
> which system are you using? (distro/wm)

Ubuntu 22.04.2 LTS with Gnome 42.5 (X11)

EpicWink, 2023-04-27:
Having the same issue with Ubuntu 22.04.2 and Gnome 42, after DKMS installing v3.2.3 (was fine on v3.2.1). I can still turn the backlight off by lowering the brightness all the way

Matombo, 2023-05-16:
I think i could reproduce it (or at least something related to it). I Suspect a gnome bug behind this: https://gitlab.gnome.org/GNOME/gnome-settings-daemon/-/issues/745

nicobo, 2023-06-27:
I'm using control center 2.0.7 and :
tuxedo-keyboard/3.2.5, 6.2.0-10007-tuxedo, x86_64: installed
tuxedo-keyboard/3.2.5, 6.2.0-10010-tuxedo, x86_64: installed

I think it happens on some actions, not alone. The timestamp is around several minutes.

Still happening today.

I'm also experiencing : https://github.com/tuxedocomputers/tuxedo-keyboard/issues/179

nicobo, 2023-06-27:
And :

BIOS 1.07.03RTR
KB/EC 1.7.02TR1

BUT let me emphasize that it worked before and I haven't touched the firmware...

Matombo, 2023-06-27:
Before the backlight was driver controlled only, now the driver is controlled by the desktop environment.

Can you name the actions that trigger the backlight turning on?

nicobo, 2023-06-27:
OK that makes sense. But I'm struggling to find out what the event is. Do you have any recommendation about where to look at (some specific gnome logs maybe ?)

closed 2023-12-11 by tuxedo-bot