tuxedocomputers/tuxedo-keyboard #179: Pulse 15 Gen 1 keyboard backlight ignores settings in bios or last state. It is always powered on while or after booting or powers itself on after a while

toastedghost, 2023-04-23:
Hi,

had postet about this issue already here https://github.com/tuxedocomputers/tuxedo-control-center/issues/280 but it seems more to be a problem with the keyboard module because it still exists after i disabled the keyboard backlight settings in TCC.

My problem is, that since the update to TuxedoControlCenter 2 and with that the keyboard module, the keyboard backlight is always active at the minimum level while or after booting, although disabled in the BIOS and set to 0% in the TCC or fn+space. Disabling the keyboard backlight settings in TCC does not change anything. The backlight is still powered on after booting.

After logging in, the backlight in the TCC must first be set to a value greater than 0% and then back to 0% to turn it off again. fn+spacebar still works to select the backlight level or turn it off. 

In the meantime since the problem exists it i noticed, that after a while the backlight is automatically powered on again.

The system used is a Pulse 15 Gen 1 with Kubuntu 22.04 LTS

Matombo, 2023-05-16:
This issue is not closed, copying my questions from the other issue:

Can't reproduce it. Given: There is a very recent firmware update that is allready applied to my test device.

What you set in the BIOS menu is ignored by the OS/Tuxedo Control Center. It should set it to the last value on shutdown during boot (design decision to have a predictabel behaviour across all our devices).

Can you help me debug the rest?

Please install the latest BIOS and EC firmware version. And see if that fixes it.
Which version of tuxedo-control-center and tuxedo-keyboard are you running?

Matombo, 2023-05-17:
Woopsy wrong account, the two comments above where made by me.

toastedghost, 2023-05-17:
- Yes, both sliders change the keyboard backlight. If i change via kde slider the TCC slider changes with a little delay but shows the same value. If i change the TCC slider, the kde slider does not change, it keeps its initial value, but the keyboard backlight changes.

- If both are set to 0 the content auf `/etc/tcc/settings` is this:

`
{
  "stateMap": {
    "power_ac": "__legacy_cool_and_breezy__",
    "power_bat": "__legacy_powersave_extreme__"
  },
  "cpuSettingsEnabled": true,
  "fanControlEnabled": true,
  "ycbcr420Workaround": [
    {
      "HDMI-A-1": false,
      "eDP-1": false
    },
    {
      "DVI-I-1": false
    },
    {
      "DVI-I-2": false
    },
    {
      "DVI-I-3": false
    },
    {
      "DVI-I-4": false
    }
  ],
  "chargingProfile": "stationary",
  "keyboardBacklightControlEnabled": true,
  "keyboardBacklightColorMode": 0,
  "keyboardBacklightBrightness": 0,
  "keyboardBacklightColor": [
    0
  ]
}
`

- `dkms status` shows me this

> evdi/1.13.1, 6.2.0-10005-tuxedo, x86_64: installed
> evdi/1.13.1, 6.2.0-10007-tuxedo, x86_64: installed
> tuxedo-keyboard/3.2.3, 6.2.0-10005-tuxedo, x86_64: installed
> tuxedo-keyboard/3.2.3, 6.2.0-10007-tuxedo, x86_64: installed
> virtualbox-guest/6.1.34, 5.15.0-10025-tuxedo, x86_64: installed
> virtualbox-guest/6.1.34, 5.15.0-10033-tuxedo, x86_64: installed
> virtualbox-guest/6.1.34, 5.15.0-10041-tuxedo, x86_64: installed
> virtualbox-guest/6.1.34, 5.15.0-10058-tuxedo, x86_64: installed

- No, i do not have any other backlight controlling software running or installed.
- No, i use kubuntus default display manager sddm

- At the moment i am using the kernel 6.2.0-10007-tuxedo

toastedghost, 2023-05-17:
No, i do not have anymore disabled the backlight control in TCC. Sometimes it is nicer to use the slider to disable the keyboard backlight after a boot then to use fn+space.

Have checked that in `/etc/tcc/settings` was `"keyboardBacklightBrightness": 0` before i rebooted just now. The backlight stays off until the glowing kubuntu logo shows while booting. At that moment the keyboard backlight is switched on.

After login i checked `/etc/tcc/settings` again and there is still  `"keyboardBacklightBrightness": 0` also the TCC and kde slider both show 0% for the keyboard backlight.

If i then set the slider in TCC to 50% and again to 0% to power off the keyboard backlight.

Matombo, 2023-05-17:
I think I found it. Pushing a fix to internal testing now, keep an eye out for the next tuxedo-keyboard update and then let me know if that fixed it.

I'm leaving this issue open until then.

toastedghost, 2023-05-25:
Hi, short feedback:

with the update of the tuxedo-keyboard to version 3.2.5, the backlight now stays off after booting as expected, but there are new problems:

- fn+spacebar now only switches the backlight on at 100% or off at 0%. Switching between the different brightness levels no longer works
- the brightness is much higher than in the old version and the difference between 50% and 100% is now minimal

I can't remember exactly, but didn't there used to be more brightness levels than 0%, 50% and 100% in older versions? Or does it just seem that way to me because the backlight was otherwise weaker.

sketchbuch, 2023-05-26:
My Gen 1 Pulse 15 used to have only 3 settings, of, on low, and on high

toastedghost, 2023-05-26:
Then maybe I am confused because of the missing "on low" now and that 50% as well as 100% are almost equally bright.

closed 2023-12-11 by tuxedo-bot