tuxedocomputers/tuxedo-keyboard #180: TUXEDO InfinityBook S 15 - Gen7 backlight doesn't work mode ufter update to tuxedo-keyboard 3.2.0

LukashonakV, 2023-05-13:
The same issue #168 

modules
```
lsmod | egrep "keyboard|tuxedo"
tuxedo_io              16384  1
tuxedo_keyboard        45056  4 clevo_acpi,tuxedo_io,uniwill_wmi,clevo_wmi
```

dmesg
```
dmesg | grep -i tuxedo
[    0.000000] Linux version 6.1.27-gentoo-r1-x86_64 (root@tuxedo) (x86_64-pc-linux-gnu-gcc (Gentoo 12.2.1_p20230428-r1 p2) 12.2.1 20230428, GNU ld (Gentoo 2.39 p6) 2.39.0) #1 SMP PREEMPT_DYNAMIC Sat May 13 11:37:22 +03 2023
[    0.000000] DMI: TUXEDO TUXEDO InfinityBook S 15/17 Gen7/NS5x_7xPU, BIOS 1.07.07RTR1 06/16/2022
[   12.488100] tuxedo_keyboard: loading out-of-tree module taints kernel.
[   12.488335] tuxedo_keyboard: unknown parameter 'mode' ignored
[   12.488337] tuxedo_keyboard: unknown parameter 'color_left' ignored
[   12.488338] tuxedo_keyboard: unknown parameter 'brightness' ignored
[   12.489250] tuxedo_keyboard: module init
[   12.506199] tuxedo_keyboard: Set keyboard backlight mode on CUSTOM
[   12.510577] tuxedo_keyboard: Set keyboard enabled to: 1
[   12.512469] input: TUXEDO Keyboard as /devices/platform/tuxedo_keyboard/input/input12
[   37.151525] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 1, new_color_code FF0000
[   42.580595] tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 2, new_color_code FF00
```

LukashonakV, 2023-05-18:
I don't see backlight interface at all
[image]
Which kernel option should I turn on to have backlight interface available
[image]

tuxedoxt, 2023-05-19:
Hello,

currently you need **at least** `Led Class` and `Led Multicolor Class` options. Other "Class" options would be recommended to allow for future changes.

LukashonakV, 2023-05-19:
Hi @tuxedoxt , even
[image]
I'm still doesn't have backlight driver
[image]
In the same time there are new errors in dmesg
```fish
dmesg | grep -i tuxedo
[    0.000000] Linux version 6.1.27-gentoo-r1-x86_64 (root@tuxedo) (x86_64-pc-linux-gnu-gcc (Gentoo 12.2.1_p20230428-r1 p2) 12.2.1 20230428, GNU ld (Gentoo 2.39 p6) 2.39.0) #1 SMP PREEMPT_DYNAMIC Fri May 19 12:00:22 +03 2023
[    0.000000] DMI: TUXEDO TUXEDO InfinityBook S 15/17 Gen7/NS5x_7xPU, BIOS 1.07.07RTR1 06/16/2022
[   14.425140] tuxedo_keyboard: loading out-of-tree module taints kernel.
[   14.425253] tuxedo_keyboard: disagrees about version of symbol led_classdev_unregister
[   14.425260] tuxedo_keyboard: Unknown symbol led_classdev_unregister (err -22)
[   14.425285] tuxedo_keyboard: disagrees about version of symbol led_classdev_register_ext
[   14.425286] tuxedo_keyboard: Unknown symbol led_classdev_register_ext (err -22)
[   14.444648] tuxedo_keyboard: disagrees about version of symbol led_classdev_unregister
[   14.444654] tuxedo_keyboard: Unknown symbol led_classdev_unregister (err -22)
[   14.444690] tuxedo_keyboard: disagrees about version of symbol led_classdev_register_ext
[   14.444692] tuxedo_keyboard: Unknown symbol led_classdev_register_ext (err -22)
[   14.508594] tuxedo_keyboard: disagrees about version of symbol led_classdev_unregister
[   14.508599] tuxedo_keyboard: Unknown symbol led_classdev_unregister (err -22)
[   14.508731] tuxedo_keyboard: disagrees about version of symbol led_classdev_register_ext
[   14.508735] tuxedo_keyboard: Unknown symbol led_classdev_register_ext (err -22)
[   14.521510] tuxedo_keyboard: disagrees about version of symbol led_classdev_unregister
[   14.521514] tuxedo_keyboard: Unknown symbol led_classdev_unregister (err -22)
[   14.521611] tuxedo_keyboard: disagrees about version of symbol led_classdev_register_ext
[   14.521613] tuxedo_keyboard: Unknown symbol led_classdev_register_ext (err -22)
[   14.539728] tuxedo_keyboard: disagrees about version of symbol led_classdev_unregister
[   14.539731] tuxedo_keyboard: Unknown symbol led_classdev_unregister (err -22)
[   14.539776] tuxedo_keyboard: disagrees about version of symbol led_classdev_register_ext
[   14.539778] tuxedo_keyboard: Unknown symbol led_classdev_register_ext (err -22)
[   14.558686] tuxedo_keyboard: disagrees about version of symbol led_classdev_unregister
[   14.558691] tuxedo_keyboard: Unknown symbol led_classdev_unregister (err -22)
[   14.558861] tuxedo_keyboard: disagrees about version of symbol led_classdev_register_ext
[   14.558864] tuxedo_keyboard: Unknown symbol led_classdev_register_ext (err -22)
[   14.592157] tuxedo_keyboard: disagrees about version of symbol led_classdev_unregister
[   14.592167] tuxedo_keyboard: Unknown symbol led_classdev_unregister (err -22)
[   14.592840] tuxedo_keyboard: disagrees about version of symbol led_classdev_register_ext
[   14.592844] tuxedo_keyboard: Unknown symbol led_classdev_register_ext (err -22)
[   14.605922] tuxedo_keyboard: disagrees about version of symbol led_classdev_unregister
[   14.605927] tuxedo_keyboard: Unknown symbol led_classdev_unregister (err -22)
[   14.606863] tuxedo_keyboard: disagrees about version of symbol led_classdev_register_ext
[   14.606869] tuxedo_keyboard: Unknown symbol led_classdev_register_ext (err -22)
[   14.625251] tuxedo_keyboard: disagrees about version of symbol led_classdev_unregister
[   14.625257] tuxedo_keyboard: Unknown symbol led_classdev_unregister (err -22)
[   14.625281] tuxedo_keyboard: disagrees about version of symbol led_classdev_register_ext
[   14.625283] tuxedo_keyboard: Unknown symbol led_classdev_register_ext (err -22)
[   14.645481] tuxedo_keyboard: disagrees about version of symbol led_classdev_unregister
[   14.645486] tuxedo_keyboard: Unknown symbol led_classdev_unregister (err -22)
[   14.645509] tuxedo_keyboard: disagrees about version of symbol led_classdev_register_ext
[   14.645511] tuxedo_keyboard: Unknown symbol led_classdev_register_ext (err -22)
[   14.668388] tuxedo_keyboard: disagrees about version of symbol led_classdev_unregister
[   14.668396] tuxedo_keyboard: Unknown symbol led_classdev_unregister (err -22)
[   14.669418] tuxedo_keyboard: disagrees about version of symbol led_classdev_register_ext
[   14.669422] tuxedo_keyboard: Unknown symbol led_classdev_register_ext (err -22)
[   14.684607] tuxedo_keyboard: disagrees about version of symbol led_classdev_unregister
[   14.684611] tuxedo_keyboard: Unknown symbol led_classdev_unregister (err -22)
[   14.684626] tuxedo_keyboard: disagrees about version of symbol led_classdev_register_ext
[   14.684628] tuxedo_keyboard: Unknown symbol led_classdev_register_ext (err -22)
```

tuxedoxt, 2023-05-19:
Hmm, it complains about version. Make sure the modules build against the exact headers generated by your custom kernel build.

LukashonakV, 2023-05-19:
Very interesting... I've got 2 pack similar modules. Can you suggest which modules are actual
[image]

LukashonakV, 2023-05-19:
Thank you @tuxedoxt . Issue is solved. I removed all modules and installed them again. Now backlight is working well. Thank you a lot

closed 2023-05-19 by LukashonakV