tuxedocomputers/tuxedo-keyboard #181: Keyboard backlight is not working after update, returning module verification failed: signature and/or required key missing - tainting kernel

rigor59, 2023-05-13:
My keyboard backlight is not workin after updating to the new version, here is the output of sudo dmesg | grep -i keyboard;
```
[    3.535023] input: AT Translated Set 2 keyboard as /devices/platform/i8042/serio0/input/input0
[    5.092449] hid-generic 0003:1EA7:0066.0002: input,hidraw1: USB HID v1.10 Keyboard [2.4G Mouse] on usb-0000:00:14.0-6/input0
[    7.917997] systemd[1]: Starting Set the console keyboard layout...
[    7.937076] systemd[1]: Finished Set the console keyboard layout.
[    7.948700] tuxedo_keyboard: loading out-of-tree module taints kernel.
[    7.948739] tuxedo_keyboard: module verification failed: signature and/or required key missing - tainting kernel
[    7.949405] tuxedo_keyboard: module init
[    8.082042] tuxedo_keyboard: Registering single zone rgb leds interface failed
[    8.082046] tuxedo_keyboard: Set keyboard backlight mode on CUSTOM
[    8.089586] tuxedo_keyboard: Set keyboard enabled to: 1
[    8.094112] input: TUXEDO Keyboard as /devices/platform/tuxedo_keyboard/input/input25
```

Matombo, 2023-05-17:
Please post Distribution and desktop environment with version numbers
Also the version of tuxedo-control-center and tuxedo-keyboard you have currently installed

onurbbr, 2023-05-20:
I have this problem too. Keyboard lights work sometimes. Sometimes it doesn't work. I'm using Tuxedo Polaris 17 Gen1.
This is dmesg output:

onur@ubuntu:~$ sudo dmesg | grep tuxedo
[sudo] password for onur: 
[    2.306449] tuxedo_keyboard: loading out-of-tree module taints kernel.
[    2.306504] tuxedo_keyboard: module verification failed: signature and/or required key missing - tainting kernel
[    2.314468] tuxedo_keyboard: module init
[    2.945127] input: TUXEDO Keyboard as /devices/platform/tuxedo_keyboard/input/input24

Kernel Version:
onur@ubuntu:~$ uname -a
Linux ubuntu 6.3.3-x64v3-xanmod1 #0~20230517.05b3a98 SMP PREEMPT_DYNAMIC Thu May 18 05:17:33 UTC  x86_64 x86_64 x86_64 GNU/Linux

OS info:
onur@ubuntu:~$ cat /etc/os-release
PRETTY_NAME="Ubuntu 22.04.2 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.2 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy

App and Module versions:
TUXEDO Control Center: 2.0.5
TUXEDO Keyboard: 3.2.3

onurbbr, 2023-05-20:
I did manually compiling and packaging. I tried by reverting the commits one by one and after removing the commit, I can control the keyboard light.

The commit is this: [Move uw universal fan check, add to feats structure ](https://github.com/tuxedocomputers/tuxedo-keyboard/commit/a0d14ebd0319f289c61dd913fd91513998ab0b8c)

Matombo, 2023-05-23:
Can you please double check that you are not running in one of these bugs?

https://gitlab.gnome.org/GNOME/gnome-settings-daemon/-/issues/743

https://gitlab.gnome.org/GNOME/gnome-settings-daemon/-/issues/744

https://gitlab.gnome.org/GNOME/gnome-settings-daemon/-/issues/745

onurbbr, 2023-06-13:
I'm sorry for the late reply. The output is as follows:

```
onur@ubuntu:~$ ls /sys/class/leds
input4::capslock  input4::numlock  input4::scrolllock  mmc0::  phy0-led  rgb:kbd_backlight
```

Btw, I tried Linux Mint and the keyboard works fine. The problem is from GNOME.

Matombo, 2023-06-14:
A rework fixing https://gitlab.gnome.org/GNOME/gnome-settings-daemon/-/issues/744 is coming, no ETA though.

For the other 2 bugs i'm still waiting for the gnome maintainers reply.

onurbbr, 2023-08-13:
Hey @Matombo. The problem seems to be fixed in Ubuntu 22.04 and Fedora 38.

Ubuntu 22.04.3, 6.2 kernel, GNOME 42
Fedora 38, 6.4.9 kernel, GNOME 44

Thanks for your hard work.

[image]

IzzySoft, 2023-11-05:
I've just ran into this issue, upgrading an Aura 15 from Linux Mint 20.3 (Cinnamon) to Mint 21. It didn't start right away, but only after we added back the Tuxedo APT-repo and applied all updates. Using Timeshift to revert (we've luckily created a snapshot right before this step) and applying the changes one-by-one, we've tracked down the culprit being `tuxedo-keyboard`: as soon as that's updated, on the next reboot, keyboard backlight goes off.

For those still affected, we helped ourselves by apt-pinning `tuxedo-keyboard` to version 3.1.4 (the version that was in use with Mint 20), to prevent accidental updates in the future. For this we've placed the following content into a file named `tuxedo-keyboard.pref` located in `/etc/apt/preferences.d/`:

```python
Package: tuxedo-keyboard
Pin: version 3.1.4*
Pin-Priority: 1000
```

So until there's an update available solving this regression, the above might help you as a temporary work-around. The "upstream fix" in gnome-settings-daemon [mentioned back in June](#issuecomment-1591011282) seems not to be coming (the linked issue was closed without a comment), so it might take a while. `tuxedo-keyboard` version 3.1.4 still seems to do its job.

closed 2023-12-11 by tuxedo-bot