tuxedocomputers/tuxedo-keyboard #183: build fails with kernel 6.4.x series

xevilstar, 2023-06-16:
Dear Maintainers

sadly the build of tuxedo-keyboard dkms fails with latest kernels
the error is

make -C /lib/modules/6.4.0-rc6/build M=/usr/src/tuxedo-keyboard-3.2.6 modules
make[1]: Entering directory '/usr/src/linux-headers-6.4.0-rc6'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Debian 13.1.0-3) 13.1.0
  You are using:           gcc (Debian 13.1.0-5) 13.1.0
  CC [M]  /usr/src/tuxedo-keyboard-3.2.6/./src/tuxedo_keyboard.o
In file included from /usr/src/tuxedo-keyboard-3.2.6/./src/tuxedo_keyboard.c:23:
/usr/src/tuxedo-keyboard-3.2.6/./src/uniwill_keyboard.h: In function ‘uniwill_keyboard_probe’:
/usr/src/tuxedo-keyboard-3.2.6/./src/uniwill_keyboard.h:1146:9: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
 1146 |         struct uniwill_device_features_t *uw_feats = uniwill_get_device_features();
      |         ^~~~~~
/usr/src/tuxedo-keyboard-3.2.6/./src/uniwill_keyboard.h: At top level:
/usr/src/tuxedo-keyboard-3.2.6/./src/uniwill_keyboard.h:789:13: warning: ‘uw_charging_profile_write_state’ defined but not used [-Wunused-function]
  789 | static void uw_charging_profile_write_state(void)
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /usr/src/tuxedo-keyboard-3.2.6/./src/clevo_wmi.o
  CC [M]  /usr/src/tuxedo-keyboard-3.2.6/./src/clevo_acpi.o
  CC [M]  /usr/src/tuxedo-keyboard-3.2.6/./src/tuxedo_io/tuxedo_io.o
In file included from ./include/linux/linkage.h:7,
                 from ./include/linux/kernel.h:17,
                 from /usr/src/tuxedo-keyboard-3.2.6/./src/tuxedo_io/tuxedo_io.c:20:
/usr/src/tuxedo-keyboard-3.2.6/./src/tuxedo_io/tuxedo_io.c: In function ‘tuxedo_io_init’:
./include/linux/export.h:27:22: error: passing argument 1 of ‘class_create’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   27 | #define THIS_MODULE (&__this_module)
      |                     ~^~~~~~~~~~~~~~~
      |                      |
      |                      struct module *
/usr/src/tuxedo-keyboard-3.2.6/./src/tuxedo_io/tuxedo_io.c:826:47: note: in expansion of macro ‘THIS_MODULE’
  826 |         tuxedo_io_device_class = class_create(THIS_MODULE, "tuxedo_io");
      |                                               ^~~~~~~~~~~
In file included from ./include/linux/device.h:31,
                 from /usr/src/tuxedo-keyboard-3.2.6/./src/tuxedo_io/tuxedo_io.c:23:
./include/linux/device/class.h:230:54: note: expected ‘const char *’ but argument is of type ‘struct module *’
  230 | struct class * __must_check class_create(const char *name);
      |                                          ~~~~~~~~~~~~^~~~
/usr/src/tuxedo-keyboard-3.2.6/./src/tuxedo_io/tuxedo_io.c:826:34: error: too many arguments to function ‘class_create’
  826 |         tuxedo_io_device_class = class_create(THIS_MODULE, "tuxedo_io");
      |                                  ^~~~~~~~~~~~
./include/linux/device/class.h:230:29: note: declared here
  230 | struct class * __must_check class_create(const char *name);
      |                             ^~~~~~~~~~~~
/usr/src/tuxedo-keyboard-3.2.6/./src/tuxedo_io/tuxedo_io.c: At top level:
/usr/src/tuxedo-keyboard-3.2.6/./src/tuxedo_io/tuxedo_io.c:55:13: warning: ‘dmi_string_in’ defined but not used [-Wunused-function]
   55 | static bool dmi_string_in(enum dmi_field f, const char *str)
      |             ^~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:252: /usr/src/tuxedo-keyboard-3.2.6/./src/tuxedo_io/tuxedo_io.o] Error 1
make[1]: *** [Makefile:2026: /usr/src/tuxedo-keyboard-3.2.6] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.4.0-rc6'
make: *** [Makefile:29: all] Error 2

Thanks for your time

xevilstar, 2023-06-16:
found some patches around for modules with the same compilation error

[https://github.com/gnif/LookingGlass/issues/1075](https://github.com/gnif/LookingGlass/issues/1075)

gshollingsworth, 2023-07-03:
As of 3.2.6, this has been fixed. The call to create_class in class.h under device in the kernel sources was changed from two parameters to one. This broke a number of external device modules. I have successfully built kernel 6.4.1 with tuxedo-keyboard 3.2.6 and it functions as well as before with kernel 6.3.x and tuxedo-keyboard 3.2.3.

I looked in the release notes, but this fix was not mentioned. I did see the fix in the source though, since I knew where to look.

memory-colors, 2023-09-20:
I just had the same issue on Debian 12, kernel 6.2.

It appears that this needs gcc11 (not the "latest", as the Github page suggests) and then it works.

closed 2023-12-11 by tuxedo-bot