tuxedocomputers/tuxedo-keyboard #185: Error in keyboard : clevo_leds.h "Default could not be before LABEL"

sleepingtux, 2023-07-28:
Hello, there is a bug to compile it with obs studio. 

It seems that you could not put an export label behind "default": 

So i just remove the default case in this fonctions and it compile now: 

int clevo_leds_suspend(struct platform_device *dev)
{
	switch (clevo_kb_backlight_type) {
	case CLEVO_KB_BACKLIGHT_TYPE_1_ZONE_RGB:
	case CLEVO_KB_BACKLIGHT_TYPE_3_ZONE_RGB:
		clevo_evaluate_set_keyboard_status(0);
		break;
	}
	return 0;
}
EXPORT_SYMBOL(clevo_leds_suspend);

int clevo_leds_resume(struct platform_device *dev)
{
	switch (clevo_kb_backlight_type) {
	case CLEVO_KB_BACKLIGHT_TYPE_1_ZONE_RGB:
	case CLEVO_KB_BACKLIGHT_TYPE_3_ZONE_RGB:
		clevo_evaluate_set_keyboard_status(1);
		break;
	}
	return 0;
}
EXPORT_SYMBOL(clevo_leds_resume);

sleepingtux, 2023-08-04:
To add more information, the build with out this modification on opensuse build server works for tumbleweed, but not for leap.

Here is the error:
/home/abuild/rpmbuild/BUILD/tuxedo-keyboard-3.2.9/obj/default/./src/clevo_leds.h: In function 'clevo_leds_suspend':
[   28s] /home/abuild/rpmbuild/BUILD/tuxedo-keyboard-3.2.9/obj/default/./src/clevo_leds.h:420:2: error: label at end of compound statement
[   28s]   default:
[   28s]   ^~~~~~~
[   28s] /home/abuild/rpmbuild/BUILD/tuxedo-keyboard-3.2.9/obj/default/./src/clevo_leds.h: In function 'clevo_leds_resume':
[   28s] /home/abuild/rpmbuild/BUILD/tuxedo-keyboard-3.2.9/obj/default/./src/clevo_leds.h:433:2: error: label at end of compound statement
[   28s]   default:
[   28s]   ^~~~~~~

munix9, 2023-08-04:
https://build.opensuse.org/package/view_file/home:munix9/tuxedo-keyboard/fix-label-at-end-of-compound-statement.patch?expand=1

sleepingtux, 2023-08-05:
Hello.
Yep works too. 
But should be corrected on the repo.

closed 2023-08-10 by tuxedoxt