tuxedocomputers/tuxedo-keyboard #186: modprobe on Tumbleweed fails

Chartman123, 2023-08-04:
I'm using a Pulse 14 Gen1 with openSUSE Tumbleweed and see the following errors in `systemctl status dkms`:

```
Aug 04 19:34:09 linux-tuxedo dkms.systemd[1627]: modprobe: ERROR: could not insert 'clevo_acpi': Key was rejected by service
Aug 04 19:34:09 linux-tuxedo dkms.systemd[1631]: modprobe: ERROR: could not insert 'clevo_wmi': Key was rejected by service
Aug 04 19:34:09 linux-tuxedo dkms.systemd[1633]: modprobe: ERROR: could not insert 'tuxedo_io': Key was rejected by service
Aug 04 19:34:09 linux-tuxedo dkms.systemd[1634]: modprobe: ERROR: could not insert 'tuxedo_keyboard': Key was rejected by service
Aug 04 19:34:09 linux-tuxedo dkms.systemd[1635]: modprobe: ERROR: could not insert 'uniwill_wmi': Key was rejected by service
```

I'm not quite sure since when the errors have started to appear, but it must be either since Kernel 6.4 or updating BIOS/ECC to the latest available versions.

Does anybody know what is causing the problems and what I could do to solve them?

sleepingtux, 2023-08-05:
Hello, it could be the secure boot option in the bios. 
Try to disable it to see if it works?

Chartman123, 2023-08-06:
@sleepingtux Yes, that indeed helped. Is there anything I can do to make it work with secure boot enabled?

sleepingtux, 2023-08-07:
Secureboot and module are not someting i know very well. :D
I think that the module have to be signed by a key if secure boot is enabled. 

Do you compile the module by your own or do you use a repo for the module? 
If you compile by your own i don t know how it could have been worked before. 
You could find over questions like:
https://github.com/tuxedocomputers/tuxedo-keyboard/issues/127

If you do not use a repo, you could try a repo for the tuxedo-keyboard see if it works with that?

Chartman123, 2023-08-07:
Ok no problem... 

I'm using the Tuxedo repo for openSUSE 15.5. The package there uses DKMS.

closed 2023-12-11 by tuxedo-bot