tuxedocomputers/tuxedo-keyboard #189: Model support for Non-Tuxedo computers

evertvorster, 2023-08-24:
Hi there!

How I got here... 
I am the proud owner of a Sager NP8875E laptop, which is a laptop made under license by Clevo. (I think)
I also run Linux exclusively and subscribe to a lot of Richard Stallman's ideas in regards with hardware and drivers.

There are some old instructions floating around on the internet for messing with clevo-xsm-wmi and adding in the detected model in dmidecode for my laptop before building the kernel modules.

'''
[evert@Evert tuxedo-keyboard-dkms]$ sudo dmidecode | grep "Product Name"
        Product Name: PD5x_7xSNC_SND_SNE
'''

Then, of course I see that the clevo-xms-wmi evolved into tuxedo_keyboard. 

Looking at the other issues reported here, I see that you limit support to Tuxedo computers only, and that is fair. 

However, if you do decide to be graceful and help someone get at the controls for a keyboard backlight for hardware that is either incredibly similar or exactly the same as what you support, it would be really appreciated. 

So, my main question is, how do you detect which laptop model is running, and would it be possible for me to modify a local copy of that source to detect my particular laptop just to see if your drivers could be adapted to make Linux more accesible on this particular manufacturer's hardware?

evertvorster, 2023-08-24:
Oh, I looked it up online, and the Tuxedo Gemini Gen2 is exactly the same laptop albeit under a different name:
https://www.tuxedocomputers.com/de/TUXEDO-Gemini-16-Gen2.tuxedo

evertvorster, 2023-10-01:
In commit  a175c55 the software checks for "TUXEDO" in the DMI strings. 

I can't really tell what the purpose of this checking is, but it causes this driver not to run all CLEVO and SAGER notebooks that has identical hardware.

This is against the spirit, if not the letter, of the GPL. 
You use all kinds of software that was donated to the Free Software community but then limit the drivers to only Tuxedo? 
Not very nice. 

I implore you to roll back this change.

goto-loop, 2023-10-04:
So that's why the software stopped working on my XMG notebook.

Is this the passion for Linux TUXEDO advertises [on their website](https://www.tuxedocomputers.com/en/why-TUXEDO.tuxedo) with the words:
> We are Linux and Open Source fans thoroughly. We make self developed drivers as well as themes and much more available to the community.

Why is "the community" now limited to customers? Please reconsider this questionable change.

evertvorster, 2023-10-04:
@goto-loop yeah, I'm not a fan either. I got a utility to change values in my smbios, and set the manufacturer to TUXEDO to allow me to use this software unmodified. Previously I just reverted this change on my local copy of the source and built it like that... but if this commit is not fixed I have a good mind to fork this repo.

vinzv, 2023-10-09:
We develop tools and drivers for our laptops sold. This includes testing and QA for these exact models. To ensure everything works for customers who buy devices from TUXEDO we added these checks. This is not to prohibit other vendor's models from using it, but to make sure everything works. 
While this may sound slightly like a generic excuse, let me explain a bit further:

1) 
We don't buy off the shelves what ODMs offer. Instead, we have quite a huge influence on what and how components are built in by ODMs. And so do others potentially as well, depending on how big they are. So assuming retailers and vendors of the same base model offer the exact same configuration is something you better not do. While it might go well in most cases there's no warranty of your device not being seriously damaged. Hence the following points therefore.

2) To give an example on potential problems: 
In the past we had various models where two exactly identical pieces with different strings set. While one machine ran perfectly fine, the other crashed right with the first start of TCC - every single time. Setting our own strings solved this problem.

3) Think of scaling
We're writing on the most low level chip, the EC. Imagine things going off just minimally and adding the value "50" to some register. If the ID of this register is different to our models and the value 50 gets added to maximum temperature instead of fan speed. The consequences should be obvious. 
While this is a very drastic example it is not totally out of scope. During development phases our devs stumble across things like these quite regularly. 

> This is against the spirit, if not the letter, of the GPL.

Against what spirit? The spirit of having full access to every piece of the source code? Or the spirit of being able to adapt the source code to your own needs? I don't think these things are even remotely touched. In fact, it's the exact spirit of shipping Linux perfectly fitting to modern hardware and enabling people to enjoy Linux without the hassle.

Consequently, it's not against the GPL. Please re-check your understanding of permissive licenses. Furthermore, the GPL ensured you are even aware of the change and enables you to adjust it by yourself.

Giving the reasons we hope you understand this decision. Either way I'm happy to discuss this whole story and help you at best.

evertvorster, 2023-10-09:
@vinzv Thank you for taking the time to reply. 

People who are interested in these drivers can be divided into three rough groups:

The first group is people who bought their hardware from Tuxedo. Fot this group of people the string check is superfluous, as the drivers are already running on the hardware it was written for. 

The second group are people who have hardware that is similar but different enough so that the drivers here could actually be harmful to their systems. So the string check could have some positive value, just because it stops people from damaging their own hardware. Also, you have no obligation to support hardware from other manufacturers, and definitely you are not expected to expend resources to do so. Of course, people would madly respect you if you do, but more about that later. 

The last group of people have hardware that is identical enough so that the drivers work without a flaw. It is this group that the string check influences negatively, for obvious reasons. Shutting out this group is also not in Tuxedo's interest. Let me explain:
 
From a marketing perspective, you can have Tuxedo's name in front of a lot of people who run linux on laptops, and that sort of targeted advertising is very expensive. 
Tuxedo can get it for free by just allowing other people to use drivers that they developed.  Guess what laptop these people are going to buy next? This marketing angle might even be a strong enough driver that you could be tempted to at least consider the people who use your drivers on their own hardware and not implement something that might damage the hardware, however slight the chance of that might be. 

Anecdotally, in a cruel twist of fate, I could not get a Tuxedo computer in my country, and I am stuck with identical hardware from Sager because they would ship to Namibia.  Hopefully when this laptop is worn out in a few years the situation will be different, and my next laptop will be a Tux. 

As a suggestion, the string check could "taint" the drivers, where the drivers detect that they are not running on non-Tuxedo hardware and then give a pop-up disclaimer with a warning, and still allow the user to use the drivers, once they click on "Agree", or something similar. This would keep a lot more people happy than the current situation. 

As for the spirit? 
You use GNU/Linux, right? 
Every tool, every bit of code in the entire ecosystem that you use to create bespoke laptops and earn some money for yourself and your employer has been made available for everybody else to use, free of charge, eternally.
There are no artificial restrictions on use based on hardware, and in part that is what the version 3 of the GPL is about. Definitely the spirit of free software is not holding someone to a contract... 

It's the spirit of giving to the community, even if you can't benefit from it directly. 

So, my friend, the spirit of Free Software is more than just allowing access - that's just the bare minimum! 

Please, reconsider this string check and see if there is a more elegant way to implement this that would benefit everybody.

closed 2023-12-11 by tuxedo-bot