tuxedocomputers/tuxedo-keyboard #191: Not working on Aura 15 Gen 1

ElementW, 2023-09-07:
Updating to commit e236eec63aa04c60cc930f2f58f41b223dfee5e2 after having been using 3.0.4 for a long time, the kernel module is now completely useless:
* The keyboard backlight color & brightness keys do not work any more (with no messages in the kernel log)
* A `/sys/class/leds/white:kbd_backlight` node is created but exposes a single-channel LED whose `brightness` does not even work and reads 0 all the time

tuxedoherbert, 2023-10-09:
Hi,

please post the output of `sudo dmidecode | grep "Product Name" && sudo dmidecode | grep "Manufacturer"`

Thanks and regards :)

ElementW, 2023-10-09:
```
Product Name: TUXEDO Aura 15 Gen1
Product Name: AURA1501
Manufacturer: TUXEDO
Manufacturer: TUXEDO Computers
Manufacturer: Notebook
Manufacturer: Advanced Micro Devices, Inc.
Option 1: String1 for Type12 Equipment Manufacturer
Option 2: String2 for Type12 Equipment Manufacturer
Option 3: String3 for Type12 Equipment Manufacturer
Option 4: String4 for Type12 Equipment Manufacturer
Manufacturer: Samsung
Module Manufacturer ID: Bank 1, Hex 0xCE
Memory Subsystem Controller Manufacturer ID: Unknown
Manufacturer: Samsung
Module Manufacturer ID: Bank 1, Hex 0xCE
Memory Subsystem Controller Manufacturer ID: Unknown
Manufacturer: -Virtual Battery 0-
```

However it somehow now works, I guess it was an issue with unloading a previous version and loading a new one on same boot; the backlight Fn keys work and KDE Plasma now displays a brightness slider in the Battery & Brightness tab.
Closing the issue.

closed 2023-10-09 by ElementW

tuxedoxt, 2023-10-10:
For reference, the driver reloads immediately on package update but the KDE functionality interacting with it does not re-enumerate after boot as far as I am aware.