tuxedocomputers/tuxedo-keyboard #200: Upstream location

Ionic, 2023-11-15:
Can you please add pointers to the upstream location or make your Gitlab project public?

For instance, I've been looking at 5c9ca122cca5932ec12d1967460519efd0d87894 and noticed that the linked issue (#79) has nothing to do with the code fixing it. Further digging revealed the merge commit a9df75b12f78e1e852e644723036d67f6abd6106, which points to a MR called `tuxedocomputers/development/packages/tuxedo-keyboard!51`, which is syntax Gitlab uses.

I kept looking and discovered https://gitlab.com/tuxedocomputers/development, but that project doesn't contain a public `tuxedo-keyboard` project. Not sure if it's private, renamed, moved to a different location or whatever happened, but I wasn't able to get the information I needed.

Please add pointers to the correct upstream location.

tuxedoxt, 2023-11-16:
Hello,

1. github is used as the official upstream location so that is still already the answer.
2. Yes the autoclose & autoreferencing is somewhat annoying when used like this. Until now not found a way to turn this feature off in github.

Could still happen that we switch to official gitlab in the future.

closed 2023-11-16 by tuxedoxt

Ionic, 2023-11-16:
The point is that, as a software engineer working on that piece of code, without the metadata/discussion provided in the linked MR and bug report, I can at best only **guess** at what it is supposed to do.

Right now, the change description could have equally been "changed something", which utterly useless. However, the information that actually explains it is contained in some other repository, but it seems to be private. Can't you make it public, at least read-only? Ideally, this kind of information should be part of the verbose commit message anyway, but I can only request better commit messages for future work. For stuff already published, at least make sure that linked information can actually be accessed, please.